qmk/qmk_firmware #10684: [Bug] Followed the Eclipse guide, but building targets won't work.

FilipParyz, 2020-10-18:
## Describe the Bug

So I'm an Eclipse user, I like it and use it at work. It works for me and I'd love to use it more with QMK.
Up to yesterday I worked with QMK in Eclipse but compiled on MSYS2 window opened simultaneously.
I wanted to merge making changes and building into one tool, that's why I followed [this Eclipse tutorial.](https://docs.qmk.fm/#/other_eclipse?id=setting-up-eclipse-for-qmk-development)
Unfortunately I can't get it to work. I've made three Build targets, one in the main QMK folder, one in the normal keyboard folder and one in my handwired keyboard folder.
None of them work and each gives different result.
Compiling in QMK with make command works perfectly. I've run 'make handwired/freoduo' countless times.

## System Information

 - Keyboard:  Freoduo
   - Revision (if applicable):  
 - Operating system: Windows 10
 - AVR GCC version: 8.3.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version: 0.9.45

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

- Building from main QMK folder:
[image]
- Building from handwired/freoduo folder:
[image]
- Both have the same configuration as here:
[image]

- Additionaly building from acr60 folder:
[image]
 - And it's config:
[image]

zvecr, 2022-08-28:
This issue has been automatically closed because it has not had any recent activity.
If this issue is still valid, re-open the issue and let us know.

closed 2022-08-28 by zvecr

transistorgrab, 2023-08-16:
@FilipParyz did you ever find a solution to this problem?
I'm having the same issue as described by you.
I can build in QMK MSYS but Eclipse just throws errors. (I'm on Windows if that makes a difference).
I suspect the tutorial is missing some specific information to point make in the right direction.

FilipParyz, 2023-09-29:
Sorry, unfortunately I moved away from Eclipse and never experienced problems with VS Code, so I was happy with that 🤷‍♂️