qmk/qmk_firmware #10717: [Feature Request] unicode_map with non qwerty layout are sent

jeromenerf, 2020-10-21:
## Describe the Bug

- given linux set to use dvorak layout
- when setting a unicode_map entry to `0x00E9`
- then calling XC to insert this unicode character, `00.9` is written 

This is because qwerty E is dvorak dot.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system: linux
 - AVR GCC version: 5.4 

 - ARM GCC version: arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 - QMK Firmware version: 18

## Additional Context

The following unicode_map uses unicode code points which seems to be sent to the OS as a string, prefixed with  UNICODE_KEY_LNX.
Linux on its side, just "translate" the input to dvorak.  

```c
const uint32_t PROGMEM unicode_map[] = {
	[Ocirc] = 0x00F4, // 0x00Y9
	[Eacute] = 0x00E9, //0x00D9

};
```

For the eacute, I can work around the translation issue by changing E for D, but how are supposed to manage the characters outside `[a-f0-9]`?

NB: 
```h
/* Here, we should send C-S-u but ... KC_U is G, so I use DV_U, hence KC_F*/
#define UNICODE_KEY_LNX RCTL(LSFT(KC_F))
```

fauxpark, 2020-10-22:
See https://github.com/qmk/qmk_firmware/pull/10658

jeromenerf, 2020-10-22:
Thanks @fauxpark, I think its better for me to switch the keymap back to qwerty `KC_` instead of `DV_`, then configure the keyboard XKB settings in linux as qwerty (same for iOS in fact): 

```
% cat /etc/udev/rules.d/99-planckez.rules
ACTION=="add", ATTRS{idVendor}=="feed", ATTRS{idProduct}=="6060", ENV{XKBMODEL}="pc105", ENV{XKBLAYOUT}="us,us", ENV{XKBVARIANT}=",dvorak", ENV{XKBOPTIONS}="compose:ralt"
% sudo udevadm control --reload-rules && sudo udevadm trigger
```

Note that I use the `us,us` and `,dvorak` trick to force the variant as default qwerty, otherwise xkb would default to my general system variant, which is `us(dvorak)`, for the laptop builtin keyboard.

closed 2020-10-22 by jeromenerf