qmk/qmk_firmware #10738: MotoSpeed CK62 Support

SmileYzn, 2020-10-23:
This have support for MotoSpeed CK62? or just can add suport to it?

SmileYzn, 2020-10-24:
One small thing is that keyboard is very similiar to TKL61WS if not identical, can be found here:

https://www.velocifiretech.com/product/m2-wireless-mechanical-keyboard/

Any suggestion?

tzarc, 2020-10-24:
As I mentioned on Discord: [Selecting an ARM MCU](https://docs.qmk.fm/#/platformdev_selecting_arm_mcu).

jpventura, 2020-12-01:
@Deckweiss @SmileYzn @tzarc is CK62 firmware fully functional (including bluetooth)?

SmileYzn, 2020-12-01:
Yes, i mean all in keyboard work as expected.
Can you describe better what you want bro?

Thanks for interest

jpventura, 2020-12-03:
I just bought a [Motospeed CK62][amazon-url] because it was _"budget mechanical keyboard for only $76.70"_ (yes, taxes are really high here).

First day at work I discovered that you have to hold <kbd>FN</kbd>+<kbd>3</kbd> for 5 seconds to enable the arrow keys, **disabling all other keys**. So all you have to do is hold <kbd>FN</kbd>+<kbd>3</kbd> for another 5 seconds and everything is back to normal :facepalm:

I had [**no idea** about the arrow key issues][youtube-url] and if I have to use [AutoKey][reddit-url] to _"solve"_ this issue, I prefer returning the product.

@SmileYzn All I wanna know if flashing this QMK firmware configuration is enough to solve this issue or if you had to replace any hardware component.

[amazon-url]: https://www.amazon.com/dp/B08JZ8VRL9/
[reddit-url]: https://www.reddit.com/r/MechanicalKeyboards/comments/co1agm/motospeed_ck62_function_key/
[youtube-url]: https://youtu.be/ReH0ujmJKiA?t=293

SmileYzn, 2020-12-03:
QMK do not support this keyboard :|

Anyway i used this script for autohotkey: https://gist.github.com/SmileYzn/e1144778afa2a5eb593e1f9f4ac81601

And now i'm using a GK61 at home, and motospeed at work.

brunomf, 2020-12-11:
What can we do to bring support for the Motospeed CK62?

Krops, 2021-02-03:
good keyboard except key mapping and bluetooth connection (prefere 2.4)

Krops, 2021-02-03:
Here is some info about controller https://www.eevblog.com/forum/microcontrollers/hfd2201kba-any-details-on-this-microcontroller/

closed 2021-07-08 by stale[bot]

MichalKonvic, 2021-11-14:
I don't have any experience in this, but I found this Reddit post [Motospeed CK62 - firmware update with layer fix!](https://www.reddit.com/r/MechanicalKeyboards/comments/o1gq23/motospeed_ck62_firmware_update_with_layer_fix/). Maybe someone could reverse engineer the [firmware update file](https://drive.google.com/drive/folders/1V_2lQnaxOfI1HgU82aSAClZm92X_Vjio) and create qmk support from it.

renatoalencar, 2022-05-20:
I also have one and I'm trying to fix this issue, at least for fun. I was able to dump the firmware using a ST-Linkv2 and OpenOCD, it turns the boot pin is pulled to ground already. If anyone would like to join the effort of reverse engineering the firmware. There's a dump for the boot loader and the 64kb flash memory.

[Archive.zip](https://github.com/qmk/qmk_firmware/files/8740576/Archive.zip)

HorrorTroll, 2022-05-20:
HFD2201KBA is just a rebranded Sonix SN32F248B MCU make by HFD company, for qmk support, there is a community port called SonixQMK

brunomf, 2022-05-20:
MotoSpeed released a new firmware that improves the mapping a little. Check
their website.

Notmynix, 2023-12-29:
Anyone know how to turn of caps lock light?

tzarc, 2023-12-29:
By asking their support, not QMK.