qmk/qmk_firmware #10755: Combos aren't working while everything else is

Merlin04, 2020-10-25:
I've been trying to get combos to work on multiple keyboards for a while. To add them, I've been following these steps:
1. Add `COMBO_ENABLE = yes` to the rules.mk file
2. Define a `TAPPING_TERM` in the config.h file (I set it to 500 for troubleshooting so I know the issue isn't that it is too low)
3. Add this code to the top of my keymap.c:
```c
#undef COMBO_COUNT
#define COMBO_COUNT 1

const uint16_t PROGMEM test_combo[] = {KC_W, KC_F, COMBO_END};
combo_t key_combos[COMBO_COUNT] = {COMBO(test_combo, KC_ESC)};
```

When I do this, everything on my keymap works correctly, except for the combos. I've tried:
 - Adding combos to multiple keyboard folders
 - Switching from direct pin to matrix
 - Using germ's fork of QMK with the `combos.def` system
 - Disabling unneeded features in my rules.mk
 - Removed everything in my keymap except for regular keycodes and only one base layer
 - Made sure my QMK repo is up to date
 - Reduced the number of combos to only one

How can I get combos to work?

phmongeau, 2020-10-25:
`COMBO_COUNT` should be defined in your `config.h` not `keymap.c`

precondition, 2020-10-25:
You're missing the combo enum and since you haven't defined any name for your combo (via the combo enum), you don't assign `COMBO(test_combo, KC_ESC)}` to anything.

A more correct code would be:
```c
enum combos { 
  W_F_ESC,
  
};

const uint16_t PROGMEM test_combo [] = { KC_W, KC_F, COMBO_END};

combo_t key_combos[COMBO_COUNT] = {
  [W_F_ESC] = COMBO(test_combo, KC_ESC),
  
};
```

Merlin04, 2020-10-25:
@phmongeau Thanks, that fixed the issue.
@precondition I was using the example from the QMK docs, once I moved the combo count it worked correctly.

closed 2020-10-25 by Merlin04

kohane27, 2023-07-03:
The reason my combod isn't working is because I wanted to map `jk` to `esc` but my `j` and `k` are on home-row mod. The following is how I fixed it:

`rules.mk`:

```c
COMBO_ENABLE = yes
```

`config.h`:

```c
#define COMBO_COUNT 1
#define COMBO_TERM 150
```

`keymap.c`:

```c
const uint16_t PROGMEM vim_escape[] = {RCTL_T(KC_J), RSFT_T(KC_K), COMBO_END};
combo_t key_combos[] = {
    COMBO(vim_escape, KC_ESC),
};
```

Reference: [mod-tap-support](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_combo.md#mod-tap-support)

Cheers!