qmk/qmk_firmware #10764: Downgrading to GCC 8.x is required, not recommended

samtuke, 2020-10-25:
The error message "We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended." indicates that GCC 8 is optional ("recommended") but is in fact enforced as a requirement (it's not obvious how to proceed with GCC 9 instead).

Therefore it would be better to reword this error to say this is a requirement instead of a recommendation, or else state how the recommendation can be overruled by the user.

This is an important distinction considerating that all current releases of major Linux distros ship GCC 9.x, and most provide no official way to downgrade to 8.x given how critical GCC is to other parts of the operating system.

fauxpark, 2020-10-25:
The error message is talking about avr-gcc, not the regular GCC. It is possible to build firmware using avr-gcc 9 and 10, however the compiled filesize is a bit larger than on 8.x. So we don't enforce a particular avr-gcc version, but we do recommend 8.x as it's the last version known to produce "good" output (as far as QMK is concerned).

samtuke, 2020-10-25:
OK, that's useful thanks. 

How can the recommended version requirement be overridden then?

fauxpark, 2020-10-26:
As I said, we don't enforce any specific version, there is no requirement.

samtuke, 2020-10-26:
So how should output like this be interpreted? `--help` doesn't seem to indicated a way to ignore recommendations

```
qmk setup
Ψ Found qmk_firmware at /home/samtuke/qmk_firmware.

Would you like to set /home/samtuke/qmk_firmware as your QMK home? [y/n] y

Ψ Wrote configuration to /home/samtuke/.config/qmk/qmk.ini
Ψ QMK Doctor is checking your environment.
Ψ Detected Linux.
Ψ QMK home: /home/samtuke/qmk_firmware
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.0
☒ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Problems detected, please fix these problems before proceeding.
```

fauxpark, 2020-10-26:
> Perhaps this indicates that qmk is in fact ready to use, but with warnings?

Yep. `qmk setup` just runs `qmk doctor` as a sanity check. You'll find you can compile and flash just fine with `qmk compile`/`qmk flash`, or even with the make command method, since compilation is not actually tied to the CLI  in any meaningful way.

samtuke, 2020-10-26:
In that case I misunderstood the error message and it's implication. Thanks for clarifying! Closing.

closed 2020-10-26 by samtuke

SalmanFarooqShiekh, 2024-04-27:
Thanks for the awesome work! 

Can you also please tell how much of a final size difference is there between the recommended 8.x version and other versions you tested?

It will help people (including me) to decide whether it is worth it for them to downgrade or not.

Again, many thanks for QMK!