qmk/qmk_firmware #10786: Drop (formerly Massdrop) ALT keyboard USB Hub issue

zizhenc, 2020-10-28:
QMK Configurator (https://config.qmk.fm) will cause Drop ALT keyboard USB hub not working.

## Describe the Bug

I am using the Drop (formerly Massdrop) high-profile ALT keyboard which has 2 usb-c ports as a USB 2.0 hub.
I found that if I use the online QMK Configurator (https://config.qmk.fm) to compile a firmware (bin file) and use QMK Toolbox to flash the firmware to my keyboard, my keyboard USB hub will be totally dead (not working). If I use the Drop online keyboard configurator (https://drop.com/mechanical-keyboards/configurator) to compile the firmware and use the QMK toolbox to flash the firmware to my keyboard, nothing wrong.

Furthermore, if you once used the QMK Configurator and QMK toolbox to flash the compiled firmware to the ALT keyboard, then the issue will not be recovered by reflashing the firmware compiled by Drop configurator through QMK toolbox. What happened is the QMK toolbox will show the success of flashing, but actually the firmware is not flashed into the keyboard.
I have to use the mdloader provided by Drop (https://github.com/Massdrop/mdloader/releases) to reflash the Drop configurator compiled firmware to recover the issue.

## System Information

 - Keyboard:  Drop ALT keyboard
   - Revision (if applicable):  
 - Operating system: macOS Catalina (10.15.7)
 - QMK Toolbox version: 0.0.20(1)
 - MCU (AVR only): atmega32u4

zizhenc, 2020-10-29:
I found that the usb hub is not totally dead, it seems using QMK configurator compiled firmware will make the USB hub only support pure USB 2.0 USB-c cable. Any usb 3.0 usb-c cable would have no power. But using Drop configurator compiled firmware has no such problem.

drashna, 2020-10-31:
This is likely because the code that massdrop is using is more up to date than the QMK repo.  

Basically, it needs to be updated on our end... however, the only ones really familiar with the code is Massdrop or whomever is maintaining their code this week.

zizhenc, 2020-10-31:
> This is likely because the code that massdrop is using is more up to date than the QMK repo.
> 
> Basically, it needs to be updated on our end... however, the only ones really familiar with the code is Massdrop or whomever is maintaining their code this week.

I thought Drop is also using QMK firmware to compile the keymap. The point is if I use Drop provided mdloader to flash the QMK configurator compiled firmware, the issue will still happen.

tzarc, 2020-10-31:
They do use QMK, but their copy has diverged from master, and configurator builds master. Massdrop has a... poor record of keeping things up to date in QMK and regularly ignores requests for cleanup before integration.

Their rocketeer PR has been sitting around for a year now, with seemingly zero interest in bringing it up to standards -- I'd envisage a similar issue in this case with the ALT's hub, should they eventually decide to submit their changes.

tzarc, 2020-10-31:
For the record, they changed their own MCU without updating mdloader for a significant length of time, so even with their own tools they seem to struggle to keep things current.

cmerrill, 2020-11-01:
@zizhenc It's likely that the needed fixes are unfortunately in the previously mentioned rocketeer PR.
It looks like the USB handling code was also updated in that PR (#6568). I'm going to try to look into grabbing the specific changes and integrating them.

zizhenc, 2020-11-01:
> @zizhenc It's likely that the needed fixes are unfortunately in the previously mentioned rocketeer PR.
> It looks like the USB handling code was also updated in that PR (#6568). I'm going to try to look into grabbing the specific changes and integrating them.

Thanks, hope we would have a solution.

cmerrill, 2020-11-07:
Hi @zizhenc, digging through that PR I managed to figure out the issue on my ALT -- it looks like Drop made a component stuffing mistake on the USB-C CC lines on the ALT hardware design. (See [here](https://github.com/qmk/qmk_firmware/pull/6568/files#diff-2901ffe72fd5aa17ee6e257d7044809aaf90d00f59859f47104292fd9c34e7c7R69)) 

I fixed it on my board/keyboard branch here (cmerrill/qmk_firmware@d799876fc35acb527c86701a95f89353278d70c7) -- I don't think it should be merged back into the mainline because it is an ALT-only fix. You should be able to cherry-pick that specific commit if you are able to compile from source, or I could help you do that if you have any trouble.

dialtone, 2021-01-04:
Implemented your patch above in my qmk local firmware but sadly it doesn't really fix it, nor it makes my external usb dac work either when plugged through the keyboard.

nrivers, 2021-09-25:
@NicktheDev has a Medium article to show you how to flash from start to finish. https://medium.com/@nickthedev/how-to-configure-flash-your-drop-alt-ctrl-carina-shift-keyboard-in-2021-148181a8554

drashna, 2024-10-13:
massdrop/drop boards removed ATM, and future versions will likely be using RIOT.

closed 2024-10-13 by drashna