qmk/qmk_firmware #10789: [Bug] Error compiling for ergodox_infinity: no rule for printf.o

focusaurus, 2020-10-29:
I have a configurator json layout I use on both an ergodox_ez and an ergodox_infinity. It compiles OK for the ergodox_ez but for ergodox_infinity `qmk compile` fails with a make error:

```
make[1]: *** No rule to make target '.build/obj_ergodox_infinity_test1/lib/printf/printf.o', needed by '.build/ergodox_infinity_test1.elf'.  Stop.
Make finished with errors
make: *** [Makefile:584: ergodox_infinity:test1] Error 1
```

## Describe the Bug

Make error during qmk compile

## System Information

 - Keyboard: ergodox infinity 
   - Revision (if applicable):  
 - Operating system: linux using docker
 - AVR GCC version: 5.4.0

 - ARM GCC version: 6.3.1

 - QMK Firmware version: git tag 0.10.46

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I have recently done a `docker pull qmkfw/base_container` (I was getting python3 pygments missing errors before doing that). Then I rebuilt my qmk docker image. I have tried also reverting my config json to a version known to compile fine with older qmk but I still get this error.

alanzchen, 2020-11-16:
Just passing by. I had the same error but this solves the problem for me: https://github.com/qmk/qmk_firmware/issues/10148

focusaurus, 2020-11-18:
Yup my submodules were out of sync when I built the docker image. Thanks, @alanzchen !

closed 2020-11-18 by focusaurus