qmk/qmk_firmware #10836: [Feature Request] Add more STM32F303 variants

yiancar, 2020-11-02:
STM32F303 comes in multiple variants.

STM32F303xC and STM32F303xB are essentially the same except for the flash size and RAM.

Currently when `MCU = STM32F303` it is assumed that its a STM32F303xC.
This can become more fine grain.

According to ST the die for F303xB and F303xC are the same but verified only up to the corresponding Flash and RAM sizes. This means that the ld scripts for F303xC are valid but when used for F303xB the user should be aware of the limitations. Everything else compiler side is the same.

As a first start, to add the F303xB variant, the only core change would be to change where in flash the eeprom is stored. This is trivial. If a more correct implementation is needed, the ld script should also be changed.

Currently chibios only has ld scripts for F303xC (same die as xB) and F303x8 (same die as x6). I would personally opt out of supporting x8 and x6 chips because of the 16kb RAM available compared to 40+kb 

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

tzarc, 2020-11-05:
Some variants don't have USB device support, anyway. F303K8T6, for example.
There's no reason why ldscripts can't be provided by keyboards in the short term -- anything with a user bootloader likely does so already. Realistically, this would just require dropping in a few more files into the common ldscripts folder that'll turn up with the new ChibiOS upgrade, feel free to @ me when it drops and we can see what we can sort out.

yiancar, 2020-11-05:
That sounds good,
Ill keep an eye out lets see who remembers first:P
So you think going with a `MCU=STM2F303xC` and `MCU=STM2F303xB`

Is the way to go? That to me seems best as it allows the ld file to be picked up as well.

yiancar, 2021-02-04:
any comments @tzarc :)

fauxpark, 2021-02-04:
See also https://github.com/qmk/qmk_firmware/pull/8179

closed 2021-07-08 by stale[bot]