qmk/qmk_firmware #10853: Eliminate the usage of rebase and the discussion of develop-master dance

Anton3, 2020-11-03:
## The current state of things

The repo can be nominally split into two parts:
- User space: keyboards and user directories. Changes to these directories are local and can't affect a lot of things. We want to allow more or less free modification of these files, as long as they don't break build
- Core: the rest. Changes to these files are global and can have unpredictable consequences. Multiple parallel modifications of Core may break build. We want to be extra careful about such changes and not roll them out immediately

PRs are categorizes by what parts of the repo they touch:
- PRs with changes to Core go mostly to `develop`
- PRs with changes to User space go mostly to `master`
- PRs with changes to both usually go to `develop`

`develop` and `master` gradually go out of sync. About once per week `develop` is rebased onto `master` so that users can get new core features, and `develop` can be tested on more-or-less up-to-date keymaps.

## Issues with the current approach

- Individual commits are lost in the process of rebasing
- Rebases cause chaos in PR's, which must be solved by their authors by squashing on top of `develop`
- The rebase itself is a tedious process

## Proposed solution

- PRs with changes to Core go to `develop`, they must not touch User space
- PRs with changes to User space go to `master`, they must not touch Core
- `master` is merged into `develop` frequently. Actually, it can be merged almost at any time
- `develop` is merged into `master` periodically, say every week

So far, everything is perfect, right? No conflicts! But sometimes breaking changes are required, that is, some PRs have to touch both Core and User space.

Such  PRs must wait until the next time develop is scheduled to merge into master. When the time comes:
- A temporary ban is put on merging PRs
- `master` is merged into `develop`
- The final CI check of the breaking PR passes
- The PR is merged into `develop`
- `develop` is then immediately merged into `master`
- The ban on merging PRs is lifted

So:
- Commit history is preserved (still squashed per PR)
- Rebase is banned, PRs are safe
- Merging `develop` into `master` becomes easier

FilipParyz, 2020-11-06:
I'm curious what @jackhumbert has to say about that approach.

kosorin, 2020-11-06:
I also wonder why it's necessary keep user code (keyboard, keymaps, ..) in this repository. IMHO it only makes a mess in PRs and issues. Maybe I'm just missing something so it's actually good to have it here.

Anton3, 2020-11-06:
@kosorin Instead of the develop-master dance, we could extract Core as a submodule. The general approach then stands the same. Breaking changes procedure is then modified as follows:
1. Two parallel PR's are prepared: one with changes to `qmk-core` and one with corresponding changes to `qmk-firmware`
2. A temporary ban is put on merging PRs to `qmk-core`
3. The final CI check of the breaking PR passes
4. The PR is merged into `qmk-core`'s `master`
5. The accompanying PR to `qmk-firmware` is edited to include bumping `qmk-core` version and other changes necessary to fix the build
6. A temporary ban is put on merging PRs to `qmk-firmware`
7. The final CI check of the bumping PR passes
8. The PR is merged
9. Bans are lifted

The modified process is more complicated, ceremonious and brittle than the original one. But breaking changes are relatively rare, and the horrors of this process won't affect regular PR authors. I'd claim that both processes are better than the current one.

Anton3, 2020-11-06:
Both Core and User space include multiple directories, so the repo can't be split into two in the foreseeable future.

tzarc, 2020-11-30:
This has already been mostly implemented — each merge to master is also merged to develop through GitHub actions, and we’ll be using merges for `develop`->` master` in the future. The same 3 month frequency would still apply, rather than your weekly suggestion. Are there any other concerns?

Anton3, 2020-11-30:
@tzarc Not really, except I wish the future comes soon 🙂

closed 2021-04-07 by stale[bot]