qmk/qmk_firmware #10872: One shot mod keys locking when tapped in succession

mcp292, 2020-11-06:
# Problem
My one shot mod keys are locking when tapped in succession with no alphanumeric characters in between. This often happens as a typo (I hit the wrong mod key for what I want to do), or if I decide to do something different midway.

# How to Reproduce
If I do `Ctrl-Alt-Alt`, `Ctrl` is locked until I long hold it.

Same for sequences like `Alt-OS-OS` where `Alt` is locked

# Keyboard
Corne keyboard

# OSM Bindings
```c
#define OSM_CTL OSM(MOD_LCTL)
#define OSM_ALT OSM(MOD_LALT)
#define OSM_OS  OSM(MOD_LGUI)
```

precondition, 2020-11-07:
This is the same issue as #3963

fuesec, 2025-12-10:
I am still experiencing this. any fix in sight?