qmk/qmk_firmware #10890: How do I send a keycode on keyboard startup?

cole-h, 2020-11-08:
I recently bought a QMK-compatible numpad, and am having trouble getting it to enable numlock when it starts. I've tried using `matrix_init_user` and `keyboard_post_init_user` to achieve this, but `tap_code` doesn't appear to work in those functions (too early?). Are there any later functions that get called exactly once that I could put this `tap_code` in?

Things I've tried that didn't work:

* ```
  void matrix_init_user(void) {
      tap_code(KC_NUMLOCK);
  }
  ```

* ```
  void keyboard_post_init_user(void) {
      tap_code(KC_NUMLOCK);
  }
  ```

* ```
  void matrix_init_user(void) {
      if (!(host_keyboard_led_state().num_lock))
          tap_code(KC_NUMLOCK);
  }
  ```

* ```
  void keyboard_post_init_user(void) {
      if (!(host_keyboard_led_state().num_lock))
          tap_code(KC_NUMLOCK);
  }
  ```

I also tried replacing `tap_code` with `register_code` + `unregister_code`, but to no avail.

fedinx, 2021-01-07:
I have the same problem.
I have tried
 if (! (host_keyboard_led_state (). num_lock)) {tap_code (KC_NUMLOCK); }
in a conditional inside
 void matrix_scan_user (void) {}
and it doesn't work either

kb-elmo, 2021-01-19:
numlock is not a state of the keyboard.
it's a system input state that does not depend on the keyboard or keypad that is plugged in.

if you want to have numlock enabled on your computer you should enable it up in your OS.

for windows take a look at this:
https://www.howtogeek.com/244606/how-to-enable-num-lock-automatically-when-your-computer-boots/

closed 2021-06-03 by cole-h

Myridium, 2021-09-26:
Having the same issue. `matrix_init_user` and `keyboard_post_init_user` just don't seem to be called. Debugging shows no printed message, but I get the message printed if I put the print statement inside `process_record_user`.

Myridium, 2021-09-26:
This worked for me:

```
void numlock_on(void) {
  led_t led_state = host_keyboard_led_state();
  bool b = led_state.num_lock;
  if (!b) {
      register_code(KC_NUMLOCK);
      unregister_code(KC_NUMLOCK);
  }
}
void keyboard_post_init_user() {
  numlock_on();
}
```