qmk/qmk_firmware #10901: [Bug] set_oneshot_mods(MOD_RSFT) sets MOD_RCTL

j-zero, 2020-11-09:
When trying to set MOD_RSFT with set_oneshot_mod, MOD_RCTL is set.

## Describe the Bug
While `set_oneshot_mods(MOD_LSFT)` works fine, `set_oneshot_mods(MOD_RSFT)` sets MOD_RCTL instead.

Code
```
void super_RSHIFT_finished (qk_tap_dance_state_t *state, void *user_data) {
  tap_state.state = get_dance_state(state);
  switch (tap_state.state) {
    case SINGLE_TAP:
        if(one_shot_shift_enabled){
            if(!one_shot_shift_on){
                set_oneshot_mods(MOD_RSFT);
            }
            else{
                clear_oneshot_mods();
            }
        }
       break;
    default:
        if(one_shot_shift_enabled)
            clear_oneshot_mods();
        break;
  }
}
```

## System Information

 - Keyboard:  kbdfans kbd75
   - Revision (if applicable):  rev2
 - Operating system: Windows 10 1909
 - AVR GCC version: avr-gcc.exe (GCC) 8.4.0
 - ARM GCC version: 8.3.1 20190703 
 - QMK Firmware version: 0.10.22
 - Any keyboard related software installed? 
   No

kosorin, 2020-11-10:
`MOD_RSFT` is not what you're looking for, Try `MOD_BIT(KC_RSFT)`

j-zero, 2020-11-11:
Well, that works. Thank you.
https://beta.docs.qmk.fm/using-qmk/software-features/one_shot_keys says MOD_RSFT.

> 
> 
> `MOD_RSFT` is not what you're looking for, Try `MOD_BIT(KC_RSFT)`

kosorin, 2020-11-11:
`MOD_*` is for `OSM(mod)` macro. 
I know it because I had same problem. Documentation wasn't very clear in this case so I checked source code...

zvecr, 2022-08-28:
#https://github.com/qmk/qmk_firmware/pull/10917

closed 2022-08-28 by zvecr