qmk/qmk_firmware #10902: [Bug]

timvisee, 2020-11-09:
## Describe the Bug

I'm getting an error during compilation/linking. My qmk installation is up-to-date as far as I can see, and my system is up-to-date as well.

```bash
$ ~/.local/bin/qmk compile
Ψ Compiling keymap with gmake ergodox_infinity:timvisee

QMK Firmware 0.10.50
Making ergodox_infinity with keymap timvisee

arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox_infinity/keymaps/timvisee/keymap.c                                     [OK]
Compiling: quantum/process_keycode/process_terminal.c                                               [OK]
Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/ergodox_infinity_timvisee.elf                                                       [ERRORS]
 |
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-abort.o): in function `abort':
 | /build/newlib-CVVEyx/newlib-3.3.0/build/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/stdlib/../../../../../../../../newlib/libc/stdlib/abort.c:59: undefined reference to `_exit'
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-signalr.o): in function `_kill_r':
 | /build/newlib-CVVEyx/newlib-3.3.0/build/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:53: undefined reference to `_kill'
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-signalr.o): in function `_getpid_r':
 | /build/newlib-CVVEyx/newlib-3.3.0/build/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:83: undefined reference to `_getpid'
 | collect2: error: ld returned 1 exit status
 |
gmake[1]: *** [tmk_core/rules.mk:306: .build/ergodox_infinity_timvisee.elf] Error 1
Make finished with errors
gmake: *** [Makefile:584: ergodox_infinity:timvisee] Error 1
```
Any ideas what is going on?

## System Information

 - Keyboard:  Ergodox Infinity
 - Operating system: Ubuntu 20.10
 - AVR GCC version: 5.4.0
 - ARM GCC version: arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
 - QMK Firmware version: 0.10.50

## Additional Context

This seems related: https://github.com/qmk/qmk_firmware/issues/9309

timvisee, 2020-11-09:
I do get warnings for missing `udev` rules, but I don't think that is relevant for this issue.

```bash
$ qmk setup
Ψ Found qmk_firmware at /home/timvisee/git/timvisee-qmk_firmware.

Would you like to set /home/timvisee/git/timvisee-qmk_firmware as your QMK home? [y/n] y

Ψ Wrote configuration to /home/timvisee/.config/qmk/qmk.ini
Ψ QMK Doctor is checking your environment.
Ψ Detected Linux.
⚠ Missing udev rules for 'atmel-dfu' boards. See https://docs.qmk.fm/#/faq_build?id=linux-udev-rules for more details.
⚠ Missing udev rules for 'kiibohd' boards. See https://docs.qmk.fm/#/faq_build?id=linux-udev-rules for more details.
⚠ Missing udev rules for 'stm32' boards. See https://docs.qmk.fm/#/faq_build?id=linux-udev-rules for more details.
⚠ Missing udev rules for 'bootloadhid' boards. See https://docs.qmk.fm/#/faq_build?id=linux-udev-rules for more details.
⚠ Missing udev rules for 'usbasploader' boards. See https://docs.qmk.fm/#/faq_build?id=linux-udev-rules for more details.
⚠ Missing udev rules for 'massdrop' boards. See https://docs.qmk.fm/#/faq_build?id=linux-udev-rules for more details.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing udev rules for 'caterina' boards. See https://docs.qmk.fm/#/faq_build?id=linux-udev-rules for more details.
Ψ QMK home: /home/timvisee/git/timvisee-qmk_firmware
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Problems detected, please fix these problems before proceeding.
```

DerGeist, 2021-01-03:
I got the same problem for another keyboard, but I could solve it. Simple set CONSOLE_ENABLE = yes and it compiled fine for me.

timvisee, 2021-01-03:
> I got the same problem for another keyboard, but I could solve it. Simple set CONSOLE_ENABLE = yes and it compiled fine for me.

This doesn't seem to solve it for me, sadly. I keep getting the same error.

DerGeist, 2021-01-03:
I tried it on QMK version 0.11.3 and the default layout compiled fine for me. Do you have the most recent version?

timvisee, 2021-01-03:
> I tried it on QMK version 0.11.3 and the default layout compiled fine for me. Do you have the most recent version?

Still doesn't seem to work, unless I'm missing something. This is version `0.11.38`.

Here's a recording: https://asciinema.org/a/2PRRXrCnNGR0vTEwk4OwnukFD

tzarc, 2021-01-03:
Have you got your code floating around on github somewhere?
Last I saw of this, someone was using `malloc`/`free` etc. in their code, which seems to be pulling in code related to syscalls.

timvisee, 2021-01-03:
It's available at my fork on https://github.com/timvisee/qmk_firmware/. I'm on the `user-timvisee` branch.

```bash
git clone https://github.com/timvisee/qmk_firmware.git qmk-timvisee
cd qmk-timvisee
git checkout user-timvisee
```

The keymap I'm using is at `keyboards/ergodox_infinity/keymaps/timvisee/keymap.c`.

DerGeist, 2021-01-04:
I have tested it with your fork. After changing CONSOLE_ENABLE = yes to no in the keyboards rules.mk the build worked fine for me. Compiler version: 10.2.0 on Manjaro. It seems to be caused by defining NO_PRINT in combination with TERMINAL_ENABLE = yes. Same for my compile on the ZSA fork which uses QMK Firmware 19 AUTO_SHIFT_ENABLE = yes and NO_PRINT defined. Same thing happens for your build when autoshift is enabled.

timvisee, 2021-01-04:
@DerGeist Thanks. It appears I set `TERMINAL_ENABLE` for Ergodox Infinity keyboards to true a year back. Removing [this](https://github.com/timvisee/qmk_firmware/blob/3f6075f69bcd03ef0c8f24b0739f25a763d01cd7/keyboards/ergodox_infinity/rules.mk#L20) line to disable the feature fixed the compilation. At least I'm able to flash my keyboard now!

Changing `CONSOLE_ENABLE` does not seem to affect the compilation for me.

This does still seem like it is a bug though. I might open a new issue for this specific issue (the compilation failure for Ergodox Infinity with `TERMINAL_ENABLE` set).

julowe, 2021-01-04:
Similar error codes when trying to compile [cKeys' Handwire101 keyboard](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ckeys/handwire_101). Removing `TERMINAL_ENABLE` (and associated terminal layer/keys) made it compile for me.

Though just running `make ckeys/handwire_101:default EXTRAFLAGS+=--specs=nosys.specs` (without making any changes to files in repo) also made it compile.

I'm not going in depth to actually fix this, as it works enough for me now - but figured I'd add any info to this issue to help you or future peoples.

DerGeist, 2021-01-05:
Same thing for the Moonlder and Planck compiling on the ZSA fork when TERMINAL_ENABLE or AUTO_SHIFT_ENABLE but in my case only if NO_PRINT is defined. In my case by setting COMMAND_ENABLE = no.

taotien, 2021-01-10:
The flags in rules file is what worked for me!

ridingqwerty, 2021-01-10:
@timvisee I just encountered this on a fresh Gentoo install.  Although it took a little elbow grease to get the toolchain working to compile any ARM keymaps without linkage errors related to hard/soft float, it was working until I added my fork.

Re: what @tzarc said, I have a silly TESTING macro that utilizes `malloc`/`free` to `snprintf` some arbitrary string or float value.  As soon as I comment this out, it builds fine.

Re: what @julowe said, if I leave the `malloc`/`free` code in, it builds fine with `EXTRAFLAGS+=--specs=nosys.specs`.

I've chosen to just leave it commented out for now as it's just not that important to me.  I'm not sure that anything changed recently as I am able to build `malloc`/`free` code just find on another computer without needing to use `EXTRAFLAGS`.

None of the other `rules.mk` suggestions made any difference for me (`COMMAND`, `CONSOLE`, etc are already disabled).

Sorry if none of this is useful to you.

taotien, 2021-01-11:
Today I tried compiling with just `CONSOLE_ENABLE = no` commented out it works.

tzarc, 2021-01-16:
Would be great if people striking this issue can have a go with #11573 applied.
Compilation _should_ succeed, currently untested on hardware -- would be great to hear if it's successful.

julowe, 2021-01-17:
Works for me!

For completeness:
Completely separate/new checkout of repo, pulled your pr, ran `make git-submodule`, then `make ckeys/handwire_101:default CTPC=yes`, flashed new bin to hardware and it worked.

tzarc, 2021-01-17:
Should now be fixed on `master` branch.

closed 2021-01-17 by tzarc