qmk/qmk_firmware #10908: Future of ChibiOS-Contrib

fishman, 2020-11-10:
I think the ChibiOS devs are busy with other things. I see a lot of people/forks all over the place and would like to consolidate some of that work. So I have a few questions:

Are there things in the qmk/ChibiOS-Contrib branch that are not part of ChibiOS? I just looked at qmk/ChibiOS and it's basically the 19.x release from 2019-07-14.

I assume most differences are in ChibiOS-Contrib is that correct?

My plan is to gather all ChibiOS-Contrib changes in one place and reach out to all devs that contributed.

tzarc, 2020-11-10:
The current version of qmk/ChibiOS-Contrib points to a commit on the 19.1.x branch, and has no modifications over and above what the upstream repository has -- it's just a snapshot.
I'm currently in the process of upgrading qmk/ChibiOS to 20.x, and that will in turn require a bump for qmk/ChibiOS-Contrib, also pointing towards the equivalent 20.x branch of that respective repository.

The official stance of QMK is that we require changes to be upstreamed into ChibiOS/ChibiOS-Contrib in order to be picked up by qmk/qmk_firmware -- the QMK organisation has no appetite to maintain patches on our fork of ChibiOS-Contrib, as we've been bitten in the past by people dropping code with the promise of support and subsequenly disappearing. (i.e. refer to the situation with arm_atsam)

At this stage, feel free to collect the modifications you intend to make usable, but they'll need to be PR'd upstream first.

tzarc, 2020-11-10:
Refer to the [Selecting an MCU](https://docs.qmk.fm/#/platformdev_selecting_arm_mcu) document for more information.

closed 2021-03-16 by stale[bot]