qmk/qmk_firmware #10957: PS2/mouse scrolling prevent correct operation of the keys

julpec, 2020-11-15:
Hello

I implemented a trackpoint on my keyboard following the instructions in the documentation.
Everything goes perfectly, I can type without any problem, quickly, and I can use the pointer.
Afterwards, I wanted to activate the scrolling function. From then on, typing became totally random. Overall, it's impossible to type quickly because a lot of keys are not sent.
Here is the new lines in my ``config.h`` :
```
#define PS2_MOUSE_USE_REMOTE_MODE
#define PS2_MOUSE_ENABLE_SCROLLING
#define PS2_MOUSE_SCROLL_BTN_MASK (1<<PS2_MOUSE_BTN_MIDDLE) /* Default */
```
As soon as I remove the second line above, typing becomes totally normal.

drashna, 2020-11-20:
You're using the interrupt version, right? 

Do you have `PS2_MOUSE_SCROLL_MASK` defined?

julpec, 2020-11-20:
No, I use the USART version.
I tried all the features found [here](https://docs.qmk.fm/#/feature_ps2_mouse?id=ps2-mouse-features). I really need this one:
```
#define PS2_MOUSE_USE_REMOTE_MODE
```
otherwise the scroll doesn't work well.

I also tried `#define PS2_MOUSE_SCROLL_MASK 0x0F` but it does not do anything.

weary-associate, 2020-12-07:
If you're using a trackpoint, I actually think you do NOT want to use the PS2_MOUSE_ENABLE_SCROLLING define.  It looks like this is intended for use with actual scrolling mice, i.e. with a physical scroll wheel.  If you set that define, QMK will expect the trackpoint to provide a 4-byte data packet, while a "normal" (non-scrolling) PS2 device only uses a 3-byte data packet.  So I think QMK ends up waiting for a fourth data byte including scroll data, which the trackpoint never sends, and this waiting slows down the matrix scanning.  I have found with my trackpoint build that to get scrolling, all you need is PS2_MOUSE_2CROLL_BTN_MASK, which defaults to middle.  I actually had to set it to zero, because I have one weird application which requires me to middle button drag things, and this did not work if SCROLL_BTN_MASK was non-zero.  However, in the end, I still have scrolling where I want it for the most part, because chrome and office both have in-app middle button scroll capability, and in fact, this works better for me than the QMK scrolling did.

Bottom line, I don't think you should have PS2_MOUSE_ENABLE_SCROLLING set for a basic trackpoint build.

closed 2021-06-03 by stale[bot]

humanplayer2, 2023-08-27:
Per @weary-associate comment, including 

    #define PS2_MOUSE_ENABLE_SCROLLING

messed things up for me. The mouse jumped around erratically, mouse buttons clicking left and right.