qmk/qmk_firmware #10965: Questions about custom matricies from a beginner

ThePreviousOne, 2020-11-17:
I tryed to use [this](https://www.instructables.com/How-to-Make-a-USB-Laptop-Keyboard-Controller)/ tutorial to make my of shelf keyboard programmable. But I've run into some problems and have some questions.

First: whats the difference between `#define COL2ROW` and `#define ROW2COL` whats the difference if? if its I/O then shouldn't we just use `#define INPUT_ROWS` / `#define INPUT_COLUMNS` and the same for output instead of the current method. I'm not complaining, just confused.

Second: how do I work with more than 16 rows/columns? Do I need a custom matrix?

Third: From that tutorial I found that I have 18 inputs and 8 outputs which I assume are rows and columns respectively, so I started trying to adapt bpiphany/frosty_flake. Am I right in assuming this?

Fourth: : I am using a Teensy3.2 (because I hate making PCB's and what I need was in the tutorial) is changing this method 
```
static matrix_row_t scan_col(void) {
    // Each of the 8 columns is read off pins as below
    //   7  6  5  4  3  2  1  0
    // ,--,--,--,--,--,--,--,--,
    // |B0|B3|B2|B1|B6|B4|B5|C7|
    // `--`--`--`--`--`--`--`--`
    return (
        (PINC&(1<<7) ? 0 : ((matrix_row_t)1<<0)) |
        (PINB&(1<<5) ? 0 : ((matrix_row_t)1<<1)) |
        (PINB&(1<<4) ? 0 : ((matrix_row_t)1<<2)) |
        (PINB&(1<<6) ? 0 : ((matrix_row_t)1<<3)) |
        (PINB&(1<<1) ? 0 : ((matrix_row_t)1<<4)) |
        (PINB&(1<<2) ? 0 : ((matrix_row_t)1<<5)) |
        (PINB&(1<<3) ? 0 : ((matrix_row_t)1<<6)) |
        (PINB&(1<<0) ? 0 : ((matrix_row_t)1<<7))
    );
}``` going to be complicated or can I just change the pin numbers?

pwlandoll, 2020-11-23:
`COL2ROW` and `ROW2COL` is indeed used for diode direction, but in a bigger sense I think it's used to determine if the row or column pins are used as the inputs vs. outputs. I don't know how that would be structured on your keyboard here, but you may be able to figure it out from the Arduino tester code.

I don't think you'll have to deal with custom matrix code, just to set which pins are rows and which are columns. You might have better luck looking at handwired keyboard configs in `keyboards/handwired` for inspiration. To me, it sounds like it would be similar to other handwired keyboards that use a Teensy, except that your switches are already wired up.

ThePreviousOne, 2020-11-29:
Sorry, guess I didnt specify `keyboards/handwidred/bpiphany/frosty_flake`. Its one of the few that supports more than 16 (0 - F) rows. Can matricies be represented as 4 digit strings ex. 1106 18th row 7th, or only 2 digits?

pwlandoll, 2020-12-04:
Maybe I'm misunderstanding what you're looking for, but I'm not sure why you would need to have custom matrix code. Unless the way that your off-the-shelf keyboard functions is different, you should be able to use something like `keyboards/handwired/daishi` and just specify the pins for inputs/outputs or rows/cols.

ThePreviousOne, 2020-12-07:
Thank you, I was misunderstanding, I hadn't seen a matrix with more than 16 rows/columns so I completely misunderstood how matrix layouts worked and just assumed I needed a custom matrix. Thankyou

closed 2020-12-07 by ThePreviousOne