qmk/qmk_firmware #10979: Make finished with errors on STM32F103

Leonezz, 2020-11-19:
Hi.I am making a keyboard with **stm32f103** MCU, and I want it to have a QMK firmware.But when I build my QMK project ,I got an error that I cant resolve myself.

the **rules.mk** file in the keyboard folder and keymap folder is like this:
```
# MCU name
MCU = STM32F103

RGB_MATRIX_ENABLE = WS2812
```
I am not sure where the problem is,the terminal outputs is like:
```
tmk_core/chibios.mk:156: lib/chibios-contrib/os/hal/boards/GENERIC_STM32_F103/board.mk: No such file or directory
make[1]: *** No rules to make goals “lib/chibios-contrib/os/hal/boards/GENERIC_STM32_F103/board.mk”。 STOP。
Make finished with errors
make: *** [Makefile:584：anton:anton] Error 1
```
I checked my qmk_firmware folder , there are no **lib/chibios-contrib/os/hal/boards/GENERIC_STM32_F103** path in my folder.
And I tried to build the example project in the **keyboards** folder , it works.

zvecr, 2020-11-22:
The difference right now is that boards using `MCU = STM32F103` are setting `BOOTLOADER = stm32duino` or providing their own board/linker files.

Leonezz, 2020-11-22:
> The difference right now is that boards using `MCU = STM32F103` are setting `BOOTLOADER = stm32duino` or providing their own board/linker files.

Thanks a lot. 
I tried to set `BOOTLOADER = stm32duino`,but it does not work.And I tried the keyboard **acheron/arctic** that also uses stm32 MCU ,I found the key may be some config files named chconf.h , halconf.h and **mcuconf.h**.Now I konw the **mcuconf.h** need to be modified by myself and it's related to my MCU configuration, but I am not sure about the other two.Are they MCU related or generic?
And about the definations in the **mcuconf.h** , where could I get them for my MCU?
Sorry for my ignorance in embedded programming and thanks for your answering.

tzarc, 2020-11-30:
Can you try adding `BOARD = STM32_F103_STM32DUINO` to your rules.mk?

Leonezz, 2020-11-30:
> Can you try adding `BOARD = STM32_F103_STM32DUINO` to your rules.mk?

thanks,now I understand the problem is I need a board file for stm32 MCU and I added board files to my project,it worked.

Leonezz, 2020-11-30:
this issue can now be closed

closed 2020-11-30 by Leonezz

h0oni, 2021-01-05:
@Leonezz I am building a tkl keyboard based on stm32f103 can you guide me through the firmware building if possible?

Leonezz, 2021-01-06:
> @Leonezz I am building a tkl keyboard based on stm32f103 can you guide me through the firmware building if possible?

sure, so what is holding you?

h0oni, 2021-01-06:
I can't find my blupill on qmk toolbox. How to flash into stm32f103? I mailed you please check!

linlin012, 2021-11-27:
I encountered the same problem, how can I solve it?

Leonezz, 2021-11-27:
> I encountered the same problem, how can I solve it?

You need board files for boards with stm32 MCUs, most of the time, the default board setting `STM32_F103_STM32DUINO` should be enough.
So try adding `BOARD=STM32_F103_STM32DUINO` to your rules.mk file, it may work.