qmk/qmk_firmware #10983: There are no default keymap in QMK configurator

XBowsTech, 2020-11-20:
After opening this link:https://config.qmk.fm/#/xbows/nature/LAYOUT

When I click LOAD DEFAULT,The browser prompts me:Sorry there is no default for the xbows/nature keyboard... yet!

Why is that?What should I do to get it to have the default keymap?

fauxpark, 2020-11-20:
The Configurator does not use the `default` keymap.

https://docs.qmk.fm/#/configurator_default_keymaps

XBowsTech, 2020-11-20:
> The Configurator does not use the `default` keymap.
> 
> https://docs.qmk.fm/#/configurator_default_keymaps

What can I do about it?

XBowsTech, 2020-11-20:
Let xbows/nature Configurator have the default keymap.

fauxpark, 2020-11-20:
Read the docs page I posted.

noroadsleft, 2020-11-20:
I'm actually planning on working on this today.

XBowsTech, 2020-11-21:
> I'm actually planning on working on this today. I already have a branch in the works.

That's great! thank you!

noroadsleft, 2020-11-23:
This should be working now.

https://config.qmk.fm/#/xbows/nature/LAYOUT

closed 2021-02-22 by fauxpark