qmk/qmk_firmware #10993: [Bug] Backlight flickering with XD84Pro (using soft PWM)

siggie0815, 2020-11-21:
## Describe the Bug
When I flash my XD84Pro keyboard with QMK I get horrible backlight flashing when backlight level is low. Especially when RGB underglow effects are active, the flickering is super annoying.
During compilation of the firmware, it tells me that Timer 1 is used for software PWM. This is what I expect as the PCB is using a non HW PWM pin for backlight. Also I assume the related issues described in #3555 and #3556 should be addressed in the code I use.

To solve the issue I tried the following:
- Use timer 3 instead of timer 1: PWM function for backlight didn't work at all, it was on or off, nothing in between.
- Use `BACKLIGHT_DRIVER = software`: this solves the issue as long as RGB underglow is turned off, as soon as I turn it on, the flickering is back.

I assume that there is some other ISR running and blocks processing of the backlight timer ISR. But as I'm quite new to QMK, I did not yet fully understand which timers are used for what and how everything works.

## System Information

 - Keyboard:  **XD84Pro**
 - Operating system: **all**
 - AVR GCC version: **5.4.0**
 - ARM GCC version: N/A
 - QMK Firmware version/git revision: 
**`0.10.53-33`
`gccb15c2d2`**
 - Any keyboard related software installed? **No**

fauxpark, 2020-11-21:
Duplicate of https://github.com/qmk/qmk_firmware/issues/5953

siggie0815, 2020-11-21:
Thanks for your fast reply! Damn, I didn't look into that one.

I'll check if that PR that solves the issue, later today.

siggie0815, 2020-11-21:
Indeed, PR #10491 solves the flickering issue as long as the RGB underglow effects are turned off, just as described in issue #5953.

closed 2020-11-22 by siggie0815