qmk/qmk_firmware #11015: YMD75 rev 3 Backlight malfunctioning

MrE3gman, 2020-11-23:
Once I flashed the qmk firmware (made by QMK Configurator) for the YMD75 rev3 and I plug it on my baclight will turn on (the leds over the switches) and it will slowly get dimmer dor about a minute or so, after that minute it will turn off completelly.

The board was working perfectly fine on the og firmware in terms of backlight and everything, right now the rgb controls on the qmk configurator functions to controll the underglow of the pcb no problems, but all the backlight options do not work whatsoever, and the board lights up at the start and dims itself off for about a minute.

System Information

 - Keyboard:  YMD75
   - Revision (if applicable):  rev3
 - Operating system: Win10 up to date (23/11)
 - QMK : From QMK Configurator, YMD75 rev 3 default layout (ansi layout, 6,25 spacebar, and 3 1u between spacebar and arrows)
 - Any keyboard related software installed? 
   - Nothing in particular

AfzalivE, 2020-12-23:
Does this PR fix this issue?
https://github.com/qmk/qmk_firmware/pull/11172

MrE3gman, 2020-12-24:
Yes it did, all is fine now, just flashed the lattest qmk and checked it.