qmk/qmk_firmware #11038: [Bug] International chars stop working on ubuntu 20.04

grunch, 2020-11-26:
Hi I'm having an issue with my keyboard and I would like to find the root of the cause, I will explain it now.

I'm need to use my keyboard in english and spanish, the issue appears with the spanish keys, **it only works in spanish if I set the spanish keyboard in ubuntu and restart the computer**, then I can use `ES_NTIL` but if I change to english and change back to spanish when I press `ES_NTIL` I get `;` instead of `ñ`, If I want to use again the `ñ` I need to reboot again the computer.

## System Information

 - Keyboard:  kyria rev1
 - Operating system: Ubuntu 20.04
 - AVR GCC version:  avr-gcc (GCC) 5.4.0
 - ARM GCC version: arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
 - QMK Firmware version: 0.10.50

precondition, 2021-07-28:
What's the output of `setxkbmap -print` in both scenarios?

grunch, 2021-07-28:
I am sorry my keyboard is not working anymore so I can't keep helping with this issue, I'm closing it

closed 2021-07-28 by grunch