qmk/qmk_firmware #11043: [Bug] qmk setup not working properly

chris-ryu, 2020-11-27:
qmk setup or qmk setup <repo>/qmk_firmware not working properly even after brew reinstall qmk

`Error: /Users/chrisryu/qmk_firmware is too old or not set up correctly!
Please update it or remove it completely before continuing.`

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system: macOS 10.15.7
 - AVR GCC version: 
zsh: command not found: avr-gcc
 - ARM GCC version: 
command not found: arm-none-eabi-gcc

 - QMK Firmware version:
fatal: Not a git repository (or any of the parent directories): .git

fauxpark, 2020-11-27:
As the error message says, you need to either update the qmk_firmware repo, or delete it, before running `qmk setup`.

chris-ryu, 2020-11-27:
> As the error message says, you need to either update the qmk_firmware repo, or delete it, before running `qmk setup`.

After deleting qmk_firmware directory

❯ qmk setup
☒ Could not find qmk_firmware!

Would you like to clone qmk/qmk_firmware to /Users/chrisryu/qmk_firmware?

Would you like to set /Users/chrisryu/qmk_firmware as your QMK home?

/usr/local/Cellar/qmk/0.0.36/libexec/bin/python: can't open file '/Users/chrisryu/qmk_firmware/bin/qmk': [Errno 2] No such file or directory

fauxpark, 2020-11-27:
You have to answer `y` to the first prompt. (It should actually be suffixed with `[y/n]` but the `qmk` formula specifies a slightly older version of the package that handles this - [I've made a PR to fix it](https://github.com/qmk/homebrew-qmk/pull/4))

chris-ryu, 2020-11-27:
Ok That's the solution.

closed 2020-11-27 by chris-ryu

Daveyr, 2020-12-03:
A fetch upstream has given me this error message too. `qmk install` and `qmk setup` don't work for me. The suggestion to update won't work as I am already up to date and I'd rather not remove and try again. Is there any other option?

Daveyr, 2020-12-03:
Please ignore. Forcing an upgrade and reinstall of qmk using the command below did the trick.
```
python3 -m pip install --upgrade --force-reinstall --user qmk
```