qmk/qmk_firmware #11068: [Bug] compiling preonic rev3 firmware

chandrahmuki, 2020-11-29:
I think i tried everything that i could imagine , sorry for my ignorance in advance.

https://dpaste.com/HXWFGV737

i tried to update gcc for arm eabi but didnt do any difference. 

 i would say that probably i have something missing or a lib to update but no clue what is it.

my keyboard is a preonic v3.

Thanks a lot for your help im a bit desperate with this one.

fauxpark, 2020-11-29:
Right now QMK has some problems with ARM GCC 10, you will need to downgrade to 8.x (or possibly 9.x).

chandrahmuki, 2020-11-29:
i was under 8.4 before and i had the same problem.

thanks for your help

fauxpark, 2020-11-29:
Looking closer at the error messages, I would guess something is wrong with the way that your ARM toolchain is built, specifically in regard to the floating point ABIs: https://stackoverflow.com/questions/9753749/arm-compilation-error-vfp-registers-used-by-executable-not-object-file

One of the answers here suggests to add `-mfloat-abi=hard -mfpu=fpv4-sp-d16`, however this is already the default: https://github.com/qmk/qmk_firmware/blob/master/tmk_core/chibios.mk#L287

Perhaps you can try adding them to `LDFLAGS` in your rules.mk.

chandrahmuki, 2020-11-29:
no unfortunately it doesn't change anything :(

pingshunhuangalex, 2021-03-21:
@chandrahmuki Not sure if you found a solution. I just got my preonic rev3, and are struggling with flashing atm with firmware compiled via `QMK Configurator` (https://github.com/qmk/qmk_firmware/issues/12308).

If by any chance you've found a solution, please shine some light with me cuz I'm really drowning here. Thanks.

fauxpark, 2021-03-21:
This is a problem with the ARM toolchain package(s) for your particular distro. I would recommend asking the maintainers of them about it.

uqs, 2021-08-11:
As someone who just got this to work somehow I might be able to help. The original link expired though, so what error are you currently getting?

FTR, GCC 11 on Arch Linux will produce a working firmware, as long as all RGB stuff is disabled.

closed 2022-08-29 by tzarc