qmk/qmk_firmware #11090: firmware successfully downloaded,but three cols of keys dones not work.

Leonezz, 2020-12-01:
Hi, now I have successfully compiled the firmware and had it downloaded to my MCU but there are three cols of keys dones not respond. 
My MCU is stm32f103c8x and my diode diretion is `ROW2COL`.It is really strange that all the other keys respond prefectly except the three cols.I firstly think the MCU chip broke so it is not proper functioning,but after I replaced another chip,the problem is still.I suspect maybe something in the hardware configuration is not right and hindered the three pins.
The bad cols are pinned to `A15,B3,B4`,and I checked the signal of these three pins through an oscilloscope, they are just low level signals.
All my matrix pinout is:
```C
#define MATRIX_COL_PINS { A6, B1, B9, B10, B11, B13, B14, B15, A8, A15, B3, B4, B5, B6, B7, B8 }
#define MATRIX_ROW_PINS { C13, C14, C15, A1, A4, A5 }
```
My board file is based on the `STM32F103C8_MINIMAL` board , I changed the `STM32HSECLK` to `12000000` and `STM32LSECLK` to `0` based on my board design.
I don't know where the problem is and if anyone knows how to solve it or have the same problem,please let me know,thanks very much.
I will give more infomation if you asked,I dont know what is needed for this problem.

elfmimi, 2020-12-07:
Those pins are JTAG debug pins.
See `9.3.5 JTAG/SWD alternate function remapping` in `RM0008 Reference manual` .

What's lacking is this line.
https://github.com/qmk/qmk_firmware/blob/3a69d688f2d9bc18a1e8a1f1951025d107c52855/platforms/chibios/STM32_F103_STM32DUINO/board/board.c#L55-L56

You may want to add the following code somewhere.
```C
#include "hal.h"

void board_init(void) {
   //JTAG-DP Disabled and SW-DP Enabled
   AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_JTAGDISABLE;
}
```
or just modifying to BOARD=STM32_F103_STM32DUINO in rules.mk should do the job.

If this solve your problem, please rewrite the title of this issue to more comprehensible one.

Leonezz, 2020-12-08:
> Those pins are JTAG debug pins.
> See `9.3.5 JTAG/SWD alternate function remapping` in `RM0008 Reference manual` .
>
> What's lacking is this line. [...]

thank you very much! this does solve the problem! 
sorry about the title, how about I change it to "Pin `A15,B3,B4` can't be used as matrix col pins unless JTAG debug is disabled"

closed 2021-06-03 by stale[bot]