qmk/qmk_firmware #11105: [Feature Request] Additional debouncing algorithms

NoOne2246, 2020-12-02:
I would like to add/fill in some of the missing debounce algorithms such as sym_eager_g as well as create asym algorithms for debounce.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Algorithms I plan to implement:
-  sym_eager_g
-  asym_eager_defer_g
-  asym_defer_eager_g
-  asym_eager_defer_pk
-  asym_defer_eager_pk

For the asym algorithms, I was wondering if having two different debounce times would be useful?
ie a shorter keydown timer and a longer keyup timer or vice versa.

elfmimi, 2020-12-04:
There is a lost context to this matter. -> [#8763 RFC Proposal to rename debounce algorithms](https://github.com/qmk/qmk_firmware/issues/8763)

NoOne2246, 2020-12-07:
I guess this could be a continuation of that topic, what is best course of action?
Reopen that topic and continue on with it or keep on this one?