qmk/qmk_firmware #11119: Massdrop requirements before next board gets merged

tzarc, 2020-12-04:
This issue tracks the items required for QMK to merge any new Massdrop keyboards. The upstream protocol ("arm_atsam") is not in a state that the QMK collaborators are comfortable with, and as such have decided to decline future PRs targeting arm_atsam until the following cleanup occurs:

* Merge [mdloader#16](https://github.com/Massdrop/mdloader/pull/16) -- this is a significant usability improvement for customers, and would prevent us having to explain why QMK doesn't persist data on Massdrop boards after poweroff -- at this point our response tends to be "QMK would work if Massdrop would support it"
* Move the actual Atmel/Microchip provided code into a lib/atsam or equivalent. The directory tmk_core/protocol/arm_atsam should be limited to the QMK-specific MCU+USB initialisation code as well as the main function, and thus is only inclusive of whatever is required to expose the vendor-supplied APIs to provide a concrete implementation of the interface expected by QMK.
* Move the non-Atmel/Microchip keyboard-specific code from the arm_atsam protocol (i.e. the usb2422, led drivers etc.) into to their respective keyboards, as these aren't guaranteed to be provided for every single SAMD51 board ever to be designed.
* Break out an I2C driver as per the i2c_master interface that the rest of QMK uses into drivers/arm_atsam, converting across the code that currently addresses things directly.

The i2c_master change will then enable the Massdrop boards to have the following transformations occur:

* Convert each board's LED matrix code to use QMK's rgbmatrix subsystem -- you're free to keep a "legacy mode" using your original implementation, but that should be moved to the keyboard directory and legacy mode should be opt-in, not opt-out. There is an ISSI3733 driver already in QMK.
* Convert any of the other I2C code in use (e.g. usb2422 initialisation) to i2c_master.

At this point we've raised [qmk_firmware#10904](https://github.com/qmk/qmk_firmware/pull/10904) to warn users of Massdrop's lack of support -- if we gain some traction on the above issues this will no longer be necessary (or get reverted, if it subsequently merged).

Please try to minimise each change to a single subset of modifications -- historically Massdrop's PRs have included many unrelated changes and nobody decides to review. It'd be great if this could be avoided in future.  It would also be preferable if there was an active contributor from Massdrop's side, as most of the PRs received seem to be abandonware once lodged -- any requested changes seem to be ignored.

tzarc, 2020-12-21:
Alternatively, feel free discuss with us the addition of RIOT-OS as a HAL instead of arm_atsam. Or anything else that has upstream support.

mmirate, 2021-11-26:
> Merge [mdloader#16](https://github.com/Massdrop/mdloader/pull/16) ...

Did the merging of https://github.com/Massdrop/mdloader/pull/62 suffice?

drashna, 2021-11-26:
> > Merge [mdloader#16](https://github.com/Massdrop/mdloader/pull/16) ...
> 
> Did the merging of [Massdrop/mdloader#62](https://github.com/Massdrop/mdloader/pull/62) suffice?

No.  That's only one of several part that are needed.  the mdloader was more of a "bare minimum".
The rest of the listed parts still need work done. However, a lot of prep work has been done to get everything in a place where the massdrop bards can be better supported

AndruePeters, 2022-02-04:
I did just get an updated from Drop customer service that they're in the process of reworking QMK support for their boards.

JustinSolstice, 2025-04-03:
Sorry for necroing this old issue, but has there been any progress or updates on this?

zvecr, 2025-04-03:
With https://github.com/qmk/qmk_firmware/pull/24337 merged, im going to close this ticket as it mostly covers steps/requirements that are no longer valid (or already complete).

The process of getting RIOT integrated is mostly complete outside it tracking the moving goalposts associated with it targeting the `develop` branch. I am in the middle of updating it in line with the last develop merge. 

Support for existing keyboards will return after the prerequisite PRs are merged. Till then, users could choose to build old tags (or use the various configurator apps and precompiled firmware).

closed 2025-04-03 by zvecr