qmk/qmk_firmware #11146: Adding Documentation on flashing WITHOUT QMK Toolbox and NIU MINI for RESET button

jiud, 2020-12-07:
### Why am I writing this

Hi,

I recently bought a niu mini and I noticed while going through the flashing procedure (not with QMK Toolbox, I am on Linux) that no specific instructions are given for making the niu mini get into flash mode.
Also flashing _without_ the Toolbox intructions were not specific, so I had to search on reddit for more info.

**So I think documentation on these should be added to the git for clarification, as the procedure is probably complicated to someone without previous experience on flashing.**

I figured these issues out after a while, so below I wrote my flashing "procedure" and the procedure to get the NIU MINI into flash mode

### Possible docs for flashing without QMK Toolbox

1. Use QMK Configurator for making the layout, install qmk package (depends on OS and distribution).

2. Save the layout in json file format.

3. Enter in command line: `qmk json2c [keymap_name.json] -o keymap.c` (in the directory you downloaded the .json file) to convert and save your newly created keymap in C program format for later compilation and flashing (this command will save the .c file in the directory your running it).

4. Clone this git (https://github.com/qmk/qmk_firmware.git) in home dir.

5. Navigate into the first directory of the git ("qmk_firmware" as of now). Locate your keyboard directory in the directory "keyboards" (in this case the folder was "niu mini") and under that the folder "keymaps". 

6. In the "keymaps" directory, make a new directory with your username (like for example your git username, I made the directory "juid").

7. Copy your created keymap.c from step 3 into the directory from step 6 (the directory named after your username).

8. Put your keyboard in flash mode (keyboard dependent).

9. Your keyboard should not be inputting any characters now, so with another keyboard, run from shell: `qmk flash -kb [your_keyboard] -km [your_username] ` (the your_keyboard and your_username should be the same as they were in the directories in previous steps). Example for my flashes: `qmk flash -kb niu_mini -km juid`

10. If all steps are followed correctly, after a short compilation and flashing time the keyboard should be ready for use with the new layout.

### Possible docs for NIU MINI flash mode 

I think these should be noted in the docs for niu mini:

First off, the RESET button is on the bottom of the PCB and it can be accessed with the hole on the bottom of the case.
To enter the flash mode:

1. Disconnect the niu mini from the computer.

2. _Press and hold_ the RESET button.

3. Connect the niu mini to the computer **while still holding the RESET button**.

4. After it is connected, release the reset button, the niu mini is now in flash mode.

Feel free to tell me if any of this is wrong (or if even all of it is), but I definitively feel that additional clarification is need on these subjects one way or another. 

Thank you for your time.

fauxpark, 2020-12-07:
The reset procedure should be the same as usual: just press the RESET button at any time, no need to disconnect and hold. As far as I can tell it's using the stock Atmel DFU bootloader.

jiud, 2020-12-07:
Is it like that on your Niu Mini? I tried clicking, double clicking, triple clicking, holding for 5, 10, 15 seconds while its on, while its off too, only the above way I described works.

fauxpark, 2020-12-07:
I don't have one; I'm just going off the fact that `BOOTLOADER` is set to `atmel-dfu` in the rules.mk.

jiud, 2020-12-09:
Hello again, perhaps I should expand on this since I found out whats the deal, probably for real this time.

The niu mini lets you flash it only once per time you power it off and on, if you unplug and replug it it will work as you described, it needs to be unplugged and replugged though to do it again. 

Thanks for your help, what about the other issue I described (with the docs)?

closed 2021-06-03 by stale[bot]