qmk/qmk_firmware #11147: Some Of My Keys Are Swapped Or Not Working

XBowsTech, 2020-12-07:
I have a problem. How can I avoid it?  This includes, but is not limited to, swapping Ctrl/Caps, disabling Gui, swapping Alt/Gui, swapping Backspace/Backslash, disabling all keys, and other behavioral modifications.

XBowsTech, 2020-12-07:
https://docs.qmk.fm/#/faq_keymap?id=some-of-my-keys-are-swapped-or-not-working
It seems to have been mentioned here, but I have not found a way to avoid it.

fauxpark, 2020-12-07:
https://docs.qmk.fm/#/feature_bootmagic?id=hotkeys
You probably have Bootmagic Full enabled. Use Space+Backspace while plugging the board in to clear the EEPROM to default, and optionally switch to Bootmagic Lite instead by setting `BOOTMAGIC_ENABLE = lite` in your rules.mk.

XBowsTech, 2020-12-08:
BOOTMAGIC_ENABLE = lite，I used it. That's right.
I don't understand why there is “Some Of My Keys Are Swapped ”，because I didn't use it.
And then, all of a sudden, it was triggered.

drashna, 2020-12-08:
EEPROM. If that got corrupted... it can/will cause this.  If you had full bootmagic enabled...

XBowsTech, 2020-12-08:
> https://docs.qmk.fm/#/feature_bootmagic?id=hotkeys
> You probably have Bootmagic Full enabled. Use Space+Backspace while plugging the board in to clear the EEPROM to default, and optionally switch to Bootmagic Lite instead by setting `BOOTMAGIC_ENABLE = lite` in your rules.mk.

BOOTMAGIC_ENABLE = lite，I used it. That's right.
Do you understand the reason for this problem? Because I didn't use this function, it was suddenly triggered.

fauxpark, 2020-12-08:
Trigger Bootmagic Lite by holding the top left key while plugging the board in, this will clear the EEPROM as well as reset the board into bootloader mode. You can then replug the board and the mappings should be fixed.

XBowsTech, 2020-12-08:
> Trigger Bootmagic Lite by holding the top left key while plugging the board in, this will clear the EEPROM as well as reset the board into bootloader mode. You can then replug the board and the mappings should be fixed.

I know this method, but I want to know how to avoid unexpected Swapped.

fauxpark, 2020-12-08:
Switching to Bootmagic Lite will avoid it. Bootmagic Full provides hotkeys which could be accidentally held down during powerup to swap/disable keys. This is why we recommend Lite over Full now, and it is enabled by default in the new keyboard template.

XBowsTech, 2020-12-09:
BOOTMAGIC_ENABLE = lite     # Virtual DIP switch configuration

我就是这么设置的，但还是出现了这个问题
That's how I set it up, but it still has this problem

closed 2021-06-03 by stale[bot]