qmk/qmk_firmware #11157: [Bug] Keycodes get shifted by one after #9940, that breaks correct indication in VIA

ruddy17, 2020-12-08:
Change in [#9940](https://github.com/qmk/qmk_firmware/pull/9940) causes keycodes after MIDI block in quantum_keycodes.h to shift by 1, which causes VIA to show wrong labels for certain keycodes, like RGB lighting controls. The aforementioned change is in the code block that is defined even when MIDI is disabled, so it influences all devices, even those that do not use MIDI.

kb-elmo, 2021-01-03:
Isn't this more a VIA problem?

If VIA development doesn't keep up with the changes in QMK it's not this projects problem.

ruddy17, 2021-01-03:
It is not that simple. This bug is not fixable on VIA side. The problem is that after this breaking PR got accepted, there are two conflicting sets of QMK keycodes  - old one, which is still running on countless devices all over the world, and new one, in those keebs that got their firmware flashed/updated recently. The VIA API does not have a way to know which version of QMK a keyboard runs, so VIA can't comply with this change. 
This implementation, where a keycode values assigned implicitly by their position in the list, is causing lot of problems and limits safe changes in the keycode set. Obviously #9940 shouldn't been merged, as any other PR that change the values of existing keycodes! I suggest it to be reverted and some work to be done to ensure all existing keycodes have their values stay the same between revisions.

Wilba6582, 2021-01-04:
This only affects user builds of firmware using QMK master. I haven't built/uploaded new firmware to the VIA website after this breaking change, and I can unbreak it in those builds (doing one now, actually).

drashna, 2021-03-15:
This is fixed now.

closed 2021-03-21 by fauxpark