qmk/qmk_firmware #11243: newb trying to flash preonic v3 but this error comes up

lilbbyblu, 2020-12-19:
[image]
not sure why it's saying this?

esbenboye, 2020-12-19:
You are (most liekely) missing a comma in your keymap.c file. Look at the last line (that is not commented out) before your `[_LOWER]` line. It should end with a comma, like this;

```
[_QWERTY] = LAYOUT_preonic_grid( \
  _______,          _______,    _______, _______, _______, _______, _______ , _______, _______, _______, _______, _______, \
  KC_TAB,           KC_Q,       KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,     KC_U,    KC_I,    KC_O,    KC_P,    KC_BSPACE,  \
  LCTL_T(KC_ESC),   KC_A,       KC_S,    KC_D,    KC_F,    KC_G,    KC_H,     KC_J,    KC_K,    KC_L,    _______, KC_ENTER, \
  KC_LSFT,          KC_GRV,     KC_X,    KC_C,    KC_V,    KC_B,    KC_N,     KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_RSFT,  \
  _______,          _______,    KC_LALT, KC_LGUI, LOWER,   KC_SPC,  KC_SPC,   RAISE,   KC_ROPT, _______, _______, _______  \
), << MISSING COMMA HERE
[_LOWER] = LAYOUT_preonic_grid( \
  _______, _______, _______, _______, _______, _______, _______, _______,       _______,    _______,       _______, _______, \
  _______, _______, DK_DQUO, _______, _______, DK_ACUT, DK_AMPR, DK_SLSH,       DK_LPRN,    DK_RPRN,       DK_EQL,  DK_QUES,  \
  _______, _______, _______, _______, _______, _______, _______, RALT(DK_SLSH), DK_LBRC,    DK_RBRC,       _______, DK_TILD, \
  _______, _______, _______, _______, _______, _______, _______, _______,       S(DK_LBRC), S(DK_RBRC),    DK_PLUS, _______, \
  _______, _______, _______, _______, _______, _______, _______, _______,       _______,    _______,       _______, _______ \
),
```

lilbbyblu, 2020-12-21:
> You are (most likely) missing a comma in your keymap.c file. Look at the

THANK YOU SO MUCH! I stayed up all night scanning for it and I just found
it. Not a missing comma but an added ampersand right after my layer 1.

closed 2020-12-23 by drashna