qmk/qmk_firmware #11264: Reactive lighting on kbd67v2

cdebotton, 2020-12-21:
I apologize if this isn't the best channel to ask this question, but I'm relatively (completely) new to customizing the qmk firmware. I was trying to find previous issues as to why, but I was wondering why all reactive lighting modes were disabled on the kbd67v2mkii hot swap board. I found that they were all explicitly disabled on these lines:

https://github.com/qmk/qmk_firmware/blob/master/keyboards/kbdfans/kbd67/mkiirgb/v2/config.h#L27

Could someone point me to information about this? Is there a hardware limitation that prevents the board from running these rgb modes?

Thanks for any help!

spidey3, 2020-12-21:
It's possible that it was done to save on firmware size, so as to fit into the available flash memory.
The code for the animations can be large...

cdebotton, 2020-12-22:
Thanks! I just enabled the lighting modes and saw that I still had a few kilobytes to spare with the rest of my configuration, compiled and installed with no issue!

closed 2020-12-22 by cdebotton