qmk/qmk_firmware #11265: make: *** No rule to make target `keebio/bdn9/rev2:default'.  Stop.

jedimilk, 2020-12-21:
Hello, I'm having issues flashing my macropad, If anyone knows how to fix this I'm all ears

nblyumberg, 2020-12-23:
@jedimilk can you paste a screenshot of your console? I just compiled it on my system and it worked fine:

> nblyumberg@grandpa nblyumberg % qmk compile -kb keebio/bdn9/rev2 -km default
> Ψ Compiling keymap with make keebio/bdn9/rev2:default
> 
> 
> QMK Firmware 0.11.15
> Making keebio/bdn9/rev2 with keymap default

Check to see if you have the bdn9/rev2 directory in your repo:
[image]

You can try to run qmk doctor to check your environment and if your build environment is old you may want to update that.

JSteeleIR, 2021-02-03:
I ran into this today trying to build custom keymap:

```
$ pwd 

/Users/me/src/qmk_firmware/keyboards/keebio/bdn9
$ tree
.
├── bdn9.c
├── bdn9.h
├── config.h
├── info.json
├── keymaps
│   ├── bcat
│   │   ├── keymap.c
│   │   └── readme.md
│   ├── brandonschlack
│   │   ├── config.h
│   │   ├── keymap.c
│   │   ├── readme.md
│   │   └── rules.mk
│   ├── codecoffeecode
│   │   ├── config.h
│   │   ├── keymap.c
│   │   └── rules.mk
│   ├── default
│   │   ├── config.h
│   │   └── keymap.c
│   ├── eosti
│   │   ├── config.h
│   │   ├── keymap.c
│   │   ├── readme.md
│   │   └── rules.mk
│   ├── ghostseven
│   │   ├── config.h
│   │   └── keymap.c
│   ├── hbbisenieks
│   │   └── keymap.c
│   ├── mousepad
│   │   ├── config.h
│   │   ├── keymap.c
│   │   ├── readme.md
│   │   └── rules.mk
│   ├── rishka
│   │   ├── config.h
│   │   ├── keymap.c
│   │   └── rules.mk
│   ├── test
│   │   ├── config.h
│   │   └── keymap.c
│   ├── via
│   │   ├── keymap.c
│   │   └── rules.mk
│   ├── vosechu-browser
│   │   └── keymap.c
│   ├── vosechu-ksp
│   │   └── keymap.c
│   ├── zbs2
│   └── zerobitsmith
│       ├── config.h
│       └── keymap.c
├── readme.md
├── rev1
│   ├── config.h
│   ├── rev1.c
│   ├── rev1.h
│   └── rules.mk
├── rev2
│   ├── chconf.h
│   ├── config.h
[... 121 lines cut ...]
If you have a fork already you can specify it like this:

    qmk setup <username>/qmk_firmware

This will create qmk_firmware in your home directory. If you'd like to change this location, use the -H flag as well:

    qmk setup -H /your/preferred/path
```

qmk doctor shows healthy:
```
$ qmk doctor

Ψ QMK Doctor is checking your environment.
Ψ Detected macOS.
Ψ QMK home: /Users/me/src/qmk_firmware
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

JSteeleIR, 2021-02-03:
I figured it out on my side. 

I have a 'user configuration folder' specified in the users/zerobitsmith, but I hadn't 

```
#include "zerobitsmith.h"
```

in my keymap.c, nor did I have a 'blank' zerobitsmith.c in the directory, either.

[This](https://github.com/JSteeleIR/qmk_firmware/commit/539f8259c62137e8c4d137478edd718a7f8f8b15) commit fixed my problem (even though my keymap still won't compile because I have bad keycodes in it. :p )

nblyumberg, 2021-02-03:
Awesome! Makes sense, I am not too familiar with userspace but working my way towards getting stuff set up there.

closed 2021-07-08 by stale[bot]