qmk/qmk_firmware #11277: Drop.inc Carina RGB Matrix port

iakobou, 2020-12-22:
This project aims to bring full QMK RGB Matrix support to the Drop.inc Carina 60% keyboard.

First and foremost, thank you to drashna, tzarc, and fauxpaux on either /r/olkb/ or Discord for many points of guidance in getting this project as far as it is.

I have forked QMK to host my project thus far: https://github.com/iakobou/qmk_firmware/tree/iakobou-carina-patch3.

I am happy to report that the firmware does compile successfully.

However, upon flashing it results in the Carina becoming inoperable with a flashing orange status light. I am afraid that it is beyond my capabilities now. I thought I'd put the link to the project here and maybe someone with better knowledge and know-how might want to pick it up in the future.

vmalloc, 2021-01-05:
I'm not 100% sure, but would it make more sense to port over the changes made in the official fork here: https://github.com/Massdrop/qmk_firmware/ ?

iakobou, 2021-01-05:
I used the official branch to make the carina directory in /keyboards/massdrop and then utilized the Massdrop fork to modify the carina.h, etc. 

It compiles but it does not run successfully.

drashna, 2021-03-02:
See #11119

closed 2021-07-08 by stale[bot]