qmk/qmk_firmware #11300: Travis build fails, local build works fine when case labels are not contants

Barabas5532, 2020-12-25:
This [build](https://travis-ci.org/github/qmk/qmk_firmware/builds/750723849) for my PR failed, but on my machine it built fine. It would be nice to have the same behaviour in both cases, perhaps adding some extra flags for the compiler would help?

I'm aware that the code there is not valid C, as switch labels must be constants. It seems newer version of GCC have some extension that makes it work?

    $ avr-gcc --version
    avr-gcc (GCC) 8.3.0
    Copyright (C) 2018 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

spidey3, 2020-12-27:
So, in this case the preprocessor sees `case RAISE_MASK:` and expands that the `case (1 << _RAISE):`. That's a const expression, because `_RAISE` is an enum, and enum values are, by definition, const. In theory, it should be OK in a `case` label.

This seems to me like it might be a bug in **avr-gcc 5.4.0** (used by Travis) that was fixed in later versions, including the **avr-gcc 8.3.0** in your local build environment. Or perhaps later versions of gcc do a better job of identifying const expressions at compile time...

Barabas5532, 2020-12-29:
I think you are looking at the wrong commit there. I fixed the issue by defining the masks using the preprocessor, which turns them into constant expressions. This works fine. On the commit that was failing the masks are defined like this:

```
const char *read_layer_state(void) {
    const uint8_t RAISE_MASK = 1 << _RAISE;
    const uint8_t LOWER_MASK = 1 << _LOWER;
    const uint8_t GAME_MASK  = 1 << _GAME;
    ...
```

[Here is a link to that commit](https://github.com/qmk/qmk_firmware/commit/be361dbb29d683037a6cd7bd5084c7f03a0f2abe#diff-5d43ef88ab6ceeead000dd40323d533eeee7cb5d64f4d0c51cb5e34df7af7e7bR148-R151)

spidey3, 2020-12-29:
In that case, see https://stackoverflow.com/questions/14069737/switch-case-error-case-label-does-not-reduce-to-an-integer-constant. The case label cannot be a variable that is evaluated at run time; it has to have known constant value at compile time.

So what changed here between compiler versions?  I think the difference is that the newer version of the compiler is being smarter about identifying that RAISE_MASK, etc. are constant variables at compile time...

closed 2021-06-03 by stale[bot]