qmk/qmk_firmware #11301: [Feature Pitch] Key Overrides

JonasGessner, 2020-12-25:
I have been working on a feature I call 'key override' over on my fork of QMK. I think it could be an interesting addition to the main repo.

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

This feature is already implemented. View the diff [here](https://github.com/JonasGessner/qmk_firmware/compare/upstream...jonas?expand=1).

The feature works as follows: You can define custom actions for arbitrary modifier+key combinations: if `modifier x` + `key y` are pressed, send `modifier w` + `key z` instead of `modifier x` + `key y`. This feature basically gives you the power of layers without having to use dedicated layer keys, allowing you to use modifier keys instead. 

A very simple example of one way I apply this feature: Pressing backspace while holding shift will send the delete key on my keyboard.

When the necessary keys are pressed (in the above example: shift + backspace), the override is 'activated' and the override key (+ optionally extra modifiers) is sent in the keyboard report (delete), while the trigger key (backspace) is removed from the keyboard report. The trigger modifiers _may_ be removed from the keyboard report upon activation of an override (customizable, in this example shift is removed from the report). The override is 'deactivated' in certain customizable situations, like when one of the trigger keys (shift or backspace) is lifted. The override key is then removed from the keyboard report and whichever keys are still held down _may_ be re-added to the keyboard report (this is customizable).

Some more examples where I use this feature:
• I have a 65% board, so the ^ key, which in the German layout is usually left of the 1, is taken up by the esc key. Hence I have an override that maps the combination ctrl + esc to ^. The shifted version of this key types °, which works with this key override too: press ctrl + shift + esc to type °.
• I have dedicated vol up/down keys, but no dedicated brightness up/down keys. Ctrl + vol up does brightness up, alt + vol up does a small increment of vol up, ctrl + alt + vol up does a small increment of brightness up. Same for vol/brightness down.
• I also have a 'next track' key, but no 'previous track' key. Ctrl + 'next track' does 'previous track'.

See below for a more complex example.

I only have one fn key on my keyboard (due to space limitations), so instead of having to limit myself to the fn key for custom actions I can use all modifier keys (and combinations thereof) to trigger custom actions/keycodes. Moreover, one might want to change the behavior of a common shortcut (e.g. cmd/ctrl+z), and can do so purely using the key override feature rather than having to install extra software on their PC OS.

Overrides have very fine grained control: You control which modifiers need to be pressed down to activate the override, you can control exactly which modifier+key combination is sent as a replacement. You control if the override is deactivated if certain other keys are pressed down while the override is active, and more.

## Difference to Combos

The two main differences to combos are the following:
1. There is no tapping term involved. The key combinations do not need to be pressed within a small period of time. Instead, since it relies on modifiers, it works like normal keyboard shortcuts: E.g. hold down control, and eventually press another key to do action x.
2. It is mainly focused on modifier + key combinations (although it can also create custom actions for a single key press without modifiers). This feature cannot be used to create a custom action for when two or more basic keys are held down simultaneously — this is what combos should be used for.

In addition, with the fine grained options for when an override is activated and deactivated, this feature really gives you maximum control over overrides in a way what combos does not.

## Implementation

This feature is already implemented in my fork, and is already quite clean. It is gated behind a rule `KEY_OVERRIDE_ENABLE`, and I tried following most conventions of other features.I have tested and refined it quite a bit over the last ~6 months, but it is certainly not perfect. I am deliberately not opening a PR because I first want to hear the reactions to this pitch.

A somewhat clean diff of mainline QMK <-> my fork can be viewed [here](https://github.com/JonasGessner/qmk_firmware/compare/upstream...jonas?expand=1). The feature is mostly implemented in the files `key_override.h/c`, with small supporting modifications in core QMK/TMK files.

The code includes a few changes in the QMK/TMK core, most notably adding a new bitfield of weak modifiers used exclusively for overrides, and a bit field of suppressed mods, which is used to eliminate modifiers that are held down from the keyboard report (see [here](https://github.com/JonasGessner/qmk_firmware/blob/jonas/tmk_core/common/action_util.c#L193)). This is to be able to replace arbitrary modifier+key combinations that are physically held down with other arbitrary modifier+key combinations part of an override.

Overrides are defined as structs that are pointed to from an array containing all active overrides. The fields of the `key_override_t` struct are well documented in the file [key_override.h](https://github.com/JonasGessner/qmk_firmware/blob/jonas/quantum/key_override.h). See below for an example.

## Examples

My use of key overrides can be seen in [my keymap here](https://github.com/JonasGessner/qmk_firmware/blob/jonas/keyboards/xd68/keymaps/jonas/jonas.c), which includes all the examples mentioned here.

Here is an example of the shift + backspace = delete override with comments:

```c
const key_override_t back_space_delete_override = {
    // Shift key needs to be down.
    .trigger_modifiers = MOD_MASK_SHIFT,
    // Has no effect in this case, see key_override.h for explanation.
    .one_mod = true,
    // When activated, the shift key will be lifted in the keyboard report (because we only want to send the delete key).
    .suppressed_mods = MOD_MASK_SHIFT,
    // Allow other keys being pressed when activating the override.
    .allows_other_keys_down_at_trigger = true,
    // Do not deactivate the override when another key is pressed while the override is active.
    .allows_other_keys_down_while_active = true,
    // If backspace is still held when the override is deactivated (i.e. shift is released), do not re-register backspace in the keyboard report.
    .reregister_trigger_after_completion = false,
    // Only activate when pressing shift, followed by backspace. Pressing shift while backspace is already held will not activate the override.
    .trigger_only_on_trigger_down = true,
    // Activate regardless of whether other modifiers are being held.
    .negative_mask = 0,
    // Which key is the trigger? backspace.
    .trigger = KC_BSPACE,
    // Which modifier + key combination is the replacement? Just the delete key.
    .replacement = KC_DELETE,
    // The override is always enabled (point to false to deactivate).
    .enabled = NULL};

// Null terminated array defines all overrides.
const key_override_t *const *key_overrides = (const key_override_t *const[]){&back_space_delete_override, NULL};
```

## Complex Example

Perhaps the most complex example is this: I use a German layout, so I have umlaut keys (ö, ä and ü) in the places where UK/US keyboards have braces/brackets. Brackets are hidden somewhere in alt + number combinations in the German layout, which can be quite inconvenient to reach when having to type many brackets while programming. I rarely use umlauts when I program, so I use key overrides to map brackets to these keys. Using the key override feature I map the ö and ä key to { and } (when no modifiers are pressed!). When shift is held these keys type [ and ]. When control is held they go back to their normal state of ö and ä, even when other modifiers are held (like shift to get Ö and Ä).

This is really hard to get right because typing a {, for example, is done by sending alt + 8 in the German layout. Hence pressing the ö key sends alt + 8 in the keyboard report. Now add the shift key and using a key override, a [ is typed, which is done by sending alt + 5 in the keyboard report. Now if the ctrl key is also held down (in addition to shift and ö already being held), the ö [... cut ...] ul implementation you might end up with ghosted stuck keys or wrong characters being typed. I actually tried implementing these umlaut -> bracket overrides manually using custom keycodes and handling them in `process_record_user`. Even after a decent amount of work I could still find edge cases where stuck or incorrect keys would be sent in the keyboard report. Using key overrides I have not encountered any seriously wrong behavior in edge cases, and the desired behavior is quite easy to define.

IsaacElenbaas, 2020-12-25:
Though the usual approach of changing your mods to a layer key and having everything on that layer transparent or `mod(key)` works, I think this would be very helpful if it works with other features (intercepting `tap_code` and the like) and not just simple user input.

Note the existence of #4795, a much smaller and similar pull. I'm not sure what the functional differences between these two are.

JonasGessner, 2020-12-25:
You're right about the possibility of having a layer with `mod(key)` for similar functionality, but these key overrides can do much more. For example doing a custom action when two modifiers are held, which already wouldn't work (or would be much harder) with the change-mod-key-to-layer-key approach. Not to mention the pain it would be to manually fill an extra layer with a `mod(xxx)` copies of a full keymap and keep it in sync.

This currently does not intercept stuff like `tap_code`, as it intercepts key presses in `process_record_quantum`.

Thanks for pointing to #4795, I did not know about this pull request. The PR you linked is in fact a bit slimmer than the key override feature, and it would probably be easier to use when first getting into it as it has fewer options. My implementation for key overrides started out in a similarly simple fashion. Having used and refined my code over time I found the need for all the small options that are built into it to _really_ make it work well even in weird edge cases.

With the simplicity of #4795 comes much less power, making many of the examples I mentioned above impossible, and having many possible bugs. It seems like it does not address many edge cases I deliberately addressed in my implementation, like making sure the modifiers in the key map stay correct after deactivation, using weak modifiers which are already used in other places, leading to conflicts and more. @sigpro already addresses a lot of issues in his review on that PR. Fixing a lot of the issues in #4795 would probably result in a code that is similar in size.

For some of the functionality that would only be possible with key overrides but not #4795:
• Controlling if an active override deactivates if another key is pressed
•`negative_mods`: 'blacklisting' certain modifiers that when pressed, deactivate an override (and prevent the override from activating, but this would be possible with a little manual work in #4795)
• Controlling whether the trigger key is re-registered on deactivation if it is still held down
• Handling cases where the necessary modifier(s) are pressed _after_ the required basic key is already being held
• Activating a new override in the same `process_record` event in which a currently active override is deactivated. Or in short: allowing overrides to activate on key-up events.

Take the example of the umlaut/bracket overrides I mentioned, and the complexity involved in correctly activating and deactivating the various overrides when ö, ä and modifiers are pressed and held randomly repeatedly. Trying to recreate something like this with #4795 would not work, starting with the fact that #4795 only activates overrides on key-down events, while activating overrides on a key-up event would be necessary too. The lax handling of deregistering/registering modifiers in #4795 would become a big issue too.

Ambiguities can also be resolved very explicitly using the key overrides feature. By ambiguities I mean a case when the necessary keys for two different overrides are pressed. By using `negative_mask` it can be made clear which override should be activated in such cases. For example, check my overrides for vol/brightness and little vol/little brightness up. There are many key combinations that could activate several overrides, but they are disambiguated using the `negative_mask`. Especially on deactivation of an override, I believe #4795 could run into major trouble when there are ambiguous overrides, since it does not keep track of _which_ override is activated.

I even went and tried the code from #4795 and could not get this simple override to work: map `shift` + `ö` to `alt` + `8`(which would type ] for me). Here is the code I tried to use to replicate my umlaut/bracket overrides:

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case DE_OE:
            return custom_keycode_on_modifiers(MOD_BIT(KC_LSFT), -1, record, A(5)) ||   // shift + ö: [
                   custom_keycode_on_modifiers(MOD_BIT(KC_LCTL), -1, record, DE_OE) ||  // ctrl + ö: ö
                   custom_keycode_on_modifiers(0, -1, record, A(8));                    // just ö: {
        case DE_AE:
            return custom_keycode_on_modifiers(MOD_BIT(KC_LSFT), -1, record, A(6)) ||  // shift + ä: ]
                   custom_keycode_on_modifiers(MOD_BIT(KC_LCTL), -1, record, DE_AE) || // ctrl + ä: ä
                   custom_keycode_on_modifiers(0, -1, record, A(9));                   // just ä: {
    }

    return true;
}
```

None of the specified overrides in the code snippet above work, and output looks something like this `Ö∫Ö∫Ö∫Ö∫Ö∫Ö∫Ö∫Ö∫Ö` including two keys being printed for one press and modifiers going crazy.

JonasGessner, 2020-12-26:
To improve the usability of key overrides, I [added a macro](https://github.com/JonasGessner/qmk_firmware/blob/jonas/quantum/key_override.h#L47) `BASIC_KEY_OVERRIDE`, which takes just three arguments (trigger modifiers, trigger basic key, replacement complex key). It expands to a key override definition with what I would consider the options that would be most commonly used. Someone just wanting to use some basic key overrides could already get quite far using just this macro. Hence, most people would not have to come to grips with all the options of a key override and could rely on this macro, while advanced users can tinker and fine-tune as much as they want by manually defining a `key_override_t` struct.

 With this macro, the example of shift + backspace = delete can be turned into a one-liner:

```c
key_override_t **key_overrides = (key_override_t *[]){ &BASIC_KEY_OVERRIDE(MOD_MASK_SHIFT, KC_BSPACE, KC_DELETE), NULL };
```

phntsm, 2021-05-22:
Really glad to see stale was removed from this and it appears to be in consideration. I've been searching for a way to swap parenthesis from zero and nine keys for about a month now when I have time, I like to have my number row start with zero instead of one and have just gotten by with having the parenthesis as is. While I have found a few different methods to accomplish this in a round-about way or give up and use space cadet parenthesis - this method looks absolutely perfect, as if I sent my thoughts back in time to the author and they knew just the thing to make it work hahaha
In any case, I don't know if commenting is appropriate in this forum but I just wanted to say that like many others referenced by the author, I would very much benefit from this and hope to see it integrated into QMK. Best of luck to @JonasGessner — I'm really stoked about this and really hope to be able to use this feature someday soon. Cheers!

drashna, 2021-07-23:
Closing this. 

This was added in #11422, which is merged in develop, and should be merged into master at the end of august.

closed 2021-07-23 by drashna