qmk/qmk_firmware #11325: implement support for x-bows ranger (kb-6) 87 keys keyboard

sss123next, 2020-12-27:
please implement support qmk support for x-bows ranger (kb-6) 87 keys keyboard

device identifies like this

```
[67794.800483] usb 3-6.4.2: new full-speed USB device number 38 using xhci_hcd
[67794.906232] usb 3-6.4.2: New USB device found, idVendor=1ea7, idProduct=0907, bcdDevice= 3.00
[67794.906235] usb 3-6.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[67794.906237] usb 3-6.4.2: Product: USB-HID Gaming Keyboard
[67794.906239] usb 3-6.4.2: Manufacturer: SEMITEK
[67794.906240] usb 3-6.4.2: SerialNumber: SN0000000001
[67795.072451] input: SEMITEK USB-HID Gaming Keyboard as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.0/0003:1EA7:0907.003F/input/input94
[67795.124632] hid-generic 0003:1EA7:0907.003F: input,hidraw1: USB HID v1.11 Keyboard [SEMITEK USB-HID Gaming Keyboard] on usb-0000:06:00.3-6.4.2/input0
[67795.130344] hid-generic 0003:1EA7:0907.0040: hiddev1,hidraw2: USB HID v1.11 Device [SEMITEK USB-HID Gaming Keyboard] on usb-0000:06:00.3-6.4.2/input1
[67795.138503] input: SEMITEK USB-HID Gaming Keyboard System Control as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.2/0003:1EA7:0907.0041/input/input95
[67795.190574] input: SEMITEK USB-HID Gaming Keyboard Consumer Control as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.2/0003:1EA7:0907.0041/input/input96
[67795.190642] input: SEMITEK USB-HID Gaming Keyboard as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.2/0003:1EA7:0907.0041/input/input97
[67795.190702] input: SEMITEK USB-HID Gaming Keyboard Mouse as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.2/0003:1EA7:0907.0041/input/input98
[67795.190777] hid-generic 0003:1EA7:0907.0041: input,hidraw7: USB HID v1.11 Keyboard [SEMITEK USB-HID Gaming Keyboard] on usb-0000:06:00.3-6.4.2/input2
```

```
Bus 003 Device 038: ID 1ea7:0907 SHARKOON Technologies GmbH Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1ea7 SHARKOON Technologies GmbH
  idProduct          0x0907 Keyboard
  bcdDevice            3.00
  iManufacturer           1 SEMITEK
  iProduct                2 USB-HID Gaming Keyboard
  iSerial                 3 SN0000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      64
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
[... 226 lines cut ...]
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

## Feature Request Type

- [x] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)

## Description

looks like other x-bows boards is already supported, i hope they shared common hardware and implementation of this one will not be very hard.

sss123next, 2020-12-27:
https://www.aliexpress.com/i/4000896606811.html - unfortunately manufacturer/reseller does not provide much info, but as i see manufacturer does participating in development, so here is hope for quick implementation of this

fauxpark, 2020-12-28:
You would need to know what the main chip on the PCB is, but given the mention of Semitek, I don't think it will be one that QMK supports. Semitek MCUs seem to have little to no documentation and I can't even find a website for them.

sss123next, 2020-12-28:
how can i get information about pcm chip ?

ogaskell, 2021-01-01:
If there's nothing on the product's site or listing, I imagine the easiest way would be to open up the board and look at the text on the chip itself. It should be the largest chip on the board.

sss123next, 2021-01-01:
i already requested some info from manufacturer, will wait some time before any attempts to disassemble, as i can see i need to remove all switches to disassemble case, i may broke some during this process, which is very undesired...

sss123next, 2021-02-16:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

fauxpark, 2021-02-16:
So the MCU on that board is a Weltrend WT59F164, which as far as I can tell is not ARM, so it will not be possible to run QMK on it.

sss123next, 2021-02-16:
it's sad, very nice board, interesting case.

sss123next, 2021-02-16:
http://www.weltrend.com.tw/en-global/product/detail/68/104/330 - chip specs

fauxpark, 2021-02-16:
Useless without a datasheet, unfortunately. And we would want someone be take up responsibility for maintaining the port in ChibiOS if it is ARM.
https://docs.qmk.fm/#/platformdev_selecting_arm_mcu?id=add-new-mcu-family

sss123next, 2021-02-16:
i am coding on C myself and interested in this, but unfortunately i have zero free time right now (
at least i can do test and check if it works/builds e.t.c., and probably do some coding at weekends.

elfmimi, 2021-02-16:
Interesting, but I'm pessimistic about availability of development environment.

Here they state "Andes N8 or N9 CPU" . by guessing, probably WT58 is N8 and WT59 is N9.
http://www.weltrend.com.tw/en-global/product/index/68/105

This must be the page for Andes N9.
http://www.andestech.com/en/products-solutions/andescore-processors/v3-n9/

sss123next, 2021-02-16:
http://www.andestech.com/wp-content/uploads/Andes-Software-Solutions-for-RISC-V.pdf

sss123next, 2021-02-16:
it's probably one more crosscompiller, i think i can handle this, but not on this week...

elfmimi, 2021-02-16:
After having some fun from detective work, it seems
nds32le-elf-***-v3.txz and nds32le-elf-***-v3j.txz support N9 core.
found here. https://github.com/andestech/nds-toolchain/releases

closed 2021-07-08 by stale[bot]

MrDuartePT, 2021-12-30:
I don't if this issue can be open again but I find the data sheet for the weltered controller, the bluethot rerceiver, my keyboard is sk68 will have to see if the same receiver for all gk/sk/kb6x keyboards and kinda find the datasheet for the led controller but is in Chinese 
[WT59F164_DataSheet_EN_V1.02220130927.pdf](https://github.com/qmk/qmk_firmware/files/7794594/WT59F164_DataSheet_EN_V1.02220130927.pdf)
[Infineon-CYW20730_Single_Chip_Bluetooth_Transceiver_for_Wireless_Input_Devices_Datasheet-AdditionalTechnicalInformation-v12_00-EN.pdf](https://github.com/qmk/qmk_firmware/files/7794595/Infineon-CYW20730_Single_Chip_Bluetooth_Transceiver_for_Wireless_Input_Devices_Datasheet-AdditionalTechnicalInformation-v12_00-EN.pdf)
RGB led controller website:[
](https://g-area.pro/sunmoon-ic-drivers/)