qmk/qmk_firmware #11419: How can I set USART2 for bluetooth?

Leonezz, 2021-01-03:
I have a board based on STM32F103MCU and a BTK-05 bluetooth HID board,I want to use USART2 to forward keycodes by BTK-05 bluetooth board. But I don't know how to operate USART in QMK.
If anyone konws how to initialize USART of STM32 and how to send bytes, please let me know, thanks a lot!

closed 2021-06-03 by stale[bot]

BillTercel, 2023-06-27:
To be honest, I want to know this question, I have not understood how to use the STM32F103 serial port.

Leonezz, 2023-06-27:
> To be honest, I want to know this question, I have not understood how to use the STM32F103 serial port.

I cannot remember for sure, but I believe their is no way you can do it with qmk API, I achieved it by modifying the underlying ChiBiOS code back then. Hope it helps.

BillTercel, 2023-06-28:
> > To be honest, I want to know this question, I have not understood how to use the STM32F103 serial port.
> 
> I cannot remember for sure, but I believe their is no way you can do it with qmk API, I achieved it by modifying the underlying ChiBiOS code back then. Hope it helps.

Thanks for the advice, I wonder if you remember the repair steps.
Keep that question in mind.