qmk/qmk_firmware #11469: RGB Lighting code doesn't compile

edz381, 2021-01-07:
I have followed the QMK instructions to add [lighting layers](https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgblight) to my keymap.c file

**Code snippet added to keymap file**
`// Define LED segments. Light LEDs 6 to 9 and 12 to 15 red when caps lock is active.
const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {6, 4, HSV_RED},       // Light 4 LEDs, starting with LED 6
    {12, 4, HSV_RED}       // Light 4 LEDs, starting with LED 12
);

// Light LEDs 9 & 10 in cyan when keyboard layer 1 is active
const rgblight_segment_t PROGMEM FN1_active_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {9,2, HSV_CYAN}
);

//Define the array of layers. Later layers take precedence
const rgblight_segment_t* const PROGMEM my_rgb_layers[] = RGBLIGHT_LAYERS_LIST(
    my_capslock_layer,
    FN1_active_layer
);

void keyboard_post_init_user(void) {
    // Enable the LED layers
    rgblight_layers = my_rgb_layers;
}

// Enable and disabling lighting layers
layer_state_t layer_state_set_user(layer_state_t state) {
    // Both layers will light up if both kb layers are active
    rgblight_set_layer_state(1, layer_state_cmp(state, 1));
    return state;
}

bool led_update_user(led_t led_state) {
    rgblight_set_layer_state(0, led_state.caps_lock);
    return true;
}
`

**Compiler error**
`ompiling: keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c                                    keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:70:7: error: type defaults to 'int' in declaration of 'rgblight_segment_t' [-Werror=implicit-int]
 const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
       ^~~~~~~~~~~~~~~~~~
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:70:34: error: expected ',' or ';' before 'my_capslock_layer'
 const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
                                  ^~~~~~~~~~~~~~~~~
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:71:20: error: expected identifier or '(' before ',' token
     {6, 4, HSV_RED},       // Light 4 LEDs, starting with LED 6
                    ^
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:73:1: error: expected identifier or '(' before ')' token
 );
 ^
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:76:7: error: type defaults to 'int' in declaration of 'rgblight_segment_t' [-Werror=implicit-int]
 const rgblight_segment_t PROGMEM FN1_active_layer[] = RGBLIGHT_LAYER_SEGMENTS(
       ^~~~~~~~~~~~~~~~~~
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:76:34: error: expected ',' or ';' before 'FN1_active_layer'
 const rgblight_segment_t PROGMEM FN1_active_layer[] = RGBLIGHT_LAYER_SEGMENTS(
                                  ^~~~~~~~~~~~~~~~
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:78:1: error: expected identifier or '(' before ')' token
 );
 ^
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:81:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
 const rgblight_segment_t* const PROGMEM my_rgb_layers[] = RGBLIGHT_LAYERS_LIST(
                         ^
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c: In function 'keyboard_post_init_user':
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:88:5: error: 'rgblight_layers' undeclared (first use in this function); did you mean 'rgblight_mode'?
     rgblight_layers = my_rgb_layers;
     ^~~~~~~~~~~~~~~
     rgblight_mode
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:88:5: note: each undeclared identifier is reported only once for each function it appears in
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:88:23: error: 'my_rgb_layers' undeclared (first use in this function); did you mean 'g_rgb_timer'?
     rgblight_layers = my_rgb_layers;
                       ^~~~~~~~~~~~~
                       g_rgb_timer
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c: In function 'layer_state_set_user':
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:94:5: error: implicit declaration of function 'rgblight_set_layer_state'; did you mean 'rgblight_decrease_sat'? [-Werror=implicit-function-declaration]
     rgblight_set_layer_state(1, layer_state_cmp(state, 1));
     ^~~~~~~~~~~~~~~~~~~~~~~~
     rgblight_decrease_sat
cc1: all warnings being treated as errors
 [ERRORS]`

A similar issue is described in #10041, I have rebased to master:
`╰─➤  git rebase origin/master                                                                                                                                                                                    
Current branch master is up to date.`

## System Information

KBD67 MKII RGB V2
 - Operating system: Mac OS Big Sur
 - AVR GCC version:  avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0
 - QMK Firmware version: 0.11.40

spidey3, 2021-01-07:
Have you added `#define RGBLIGHT_LAYERS` to your `config.h`?

edz381, 2021-01-07:
I have just added it and the compiler is still failing with the above errors.

spidey3, 2021-01-07:
Is your firmware directory on GitHub? 
It's very hard to diagnose this without access...

edz381, 2021-01-07:
are you able to see that PR?

edz381, 2021-01-07:
I have included RGBLIGHT_ENABLE = yes in my rules.mk and this has removed most of the compile errors (the board rules.mk file has RGB_MATRIX_ENABLE = yes which I thought worked the same as RGBLIGHT_ENABLE).

The remaining error is:
```mpiling: keyboards/kbdfans/kbd67/mkiirgb/mkiirgb.c                                               In file included from quantum/quantum.h:41,
                 from keyboards/kbdfans/kbd67/mkiirgb/mkiirgb.h:3,
                 from keyboards/kbdfans/kbd67/mkiirgb/mkiirgb.c:1:
quantum/rgblight.h:232:21: error: 'RGBLED_NUM' undeclared here (not in a function); did you mean 'RGB_HUD'?
 extern LED_TYPE led[RGBLED_NUM];
                     ^~~~~~~~~~
                     RGB_HUD
 [ERRORS]```

closed 2021-01-07 by edz381

spidey3, 2021-01-07:
> The board rules.mk file has RGB_MATRIX_ENABLE = yes which I thought worked the same as RGBLIGHT_ENABLE, but clearly not.

RGB_MATRIX_ENABLE is for the per-key RGB leds, if you have them installed. 
RGBLIGHT_ENABLE is for the underglow leds.

X-w-j9, 2024-03-22:
May I ask if your problem has been resolved? I have been encountering this problem recently, and I don't know how to handle it