qmk/qmk_firmware #11509: Error opening my .hex file

PPMAN123, 2021-01-11:
When I try to flash my pcb, It gives this error message:
`>>> dfu-programmer.exe atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Empty.
>>> dfu-programmer.exe atmega32u4 flash --force "C:\Users\Ethan Zhou\Downloads\dztech_dz65rgb_v2_default_3c4ddde8.hex"
    Error opening C:\Users\Ethan Zhou\Downloads\dztech_dz65rgb_v2_default_3c4ddde8.hex
    See --debug=51 or greater for more information.
>>> dfu-programmer.exe atmega32u4 reset`
Can anyone help?

elfmimi, 2021-01-11:
the log message you copy&pasted must be from QMK-Toolbox. if so then this should go [there](https://github.com/qmk/qmk_toolbox/issues) instead.

what does the log say in few lines above that portion?
do you see this?
```
*** Downloading the file: https://qmk.fm/compiled/dztech_dz65rgb_v2_default.hex
    File saved to: C:\Users\Ethan Zhou\Downloads\dztech_dz65rgb_v2_default_3c4ddde8.hex
```
Please check whether the file is actually there at C:\Users\Ethan Zhou\Downloads .

PPMAN123, 2021-01-12:
No, but I tried it the next morning and my issue was resolved. Thank you for helping though! :)

closed 2021-01-12 by PPMAN123