qmk/qmk_firmware #11512: [Bug] Can't get mod-tap to work on KBD75v1 for { } brackets

wkoszek, 2021-01-11:
Mod-tapping for keys `{` and `}` doesn't work.

Reading: https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/mod_tap it's unclear if it's possible to get this to work. Is  KC_LCBR = KC_LBRC + SHIFT ? If so, looks like Caveat section lists this limitation. 

## Describe the Bug

I have KBD75v1 board and I'm on macOS. Everything is functional on this board and it physically works with default keymap. 

I'm giving mod-tapping a shot. The idea is that:
- left and right SHIFT are `(` and `)` respectively (**works**)
- left and right CTRL are `[` and `]` respectively (**works**)
- left and right ALT are '{` and `}` respectively (**doesn't work**) When I press Left ALT I get `[` and Right ALT is `]`.

Looks like mod-tapping ALT isn't working with `}` and `{`. 

This is my `keymap.c`:

```
#include QMK_KEYBOARD_H

/* THIS FILE WAS GENERATED!
 *
 * This file was generated by qmk json2c. You may or may not want to
 * edit it directly.
 */

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
	[0] = LAYOUT(
	    KC_ESC, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_F12, KC_PSCR, KC_SLCK, KC_PAUS,
	    KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_NO, KC_BSPC, KC_HOME,
	    KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSLS, KC_PGUP,
	    OSM(MOD_HYPR), KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_PGDN,
	    KC_LSPO, KC_NO, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSPC, KC_UP, KC_END,
	    LCTL_T(KC_LBRC), LALT_T(KC_LCBR), LGUI(KC_NO), KC_SPC, KC_SPC, KC_SPC, MO(1), RALT_T(KC_RCBR), RCTL_T(KC_RBRC), KC_LEFT, KC_DOWN, KC_RGHT
	),
	[1] = LAYOUT(KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_MRWD, KC_MPRV, KC_MPLY, KC_MNXT, KC_MFFD, KC_MUTE, KC_VOLD, KC_VOLU, KC_NO, KC_NO, KC_EJCT, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RESET, KC_TRNS, KC_TRNS, RGB_TOG, RGB_MOD, RGB_HUI, RGB_HUD, RGB_SAI, RGB_SAD, KC_TRNS, KC_MS_U, KC_WH_U, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_VAI, RGB_VAD, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_MS_L, KC_MS_D, KC_MS_R, KC_WH_D, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, BL_DEC, BL_TOGG, BL_INC, BL_STEP, KC_BTN1, KC_BTN2, KC_BTN3, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS)
};
```

Expected is to have LALT generate {, but getting [. For RALT I'm getting ] as opposed to }

## System Information

 - Keyboard:  KB75v1
   - Revision (if applicable):   rev 1
 - Operating system: macOS Catalina
 - AVR GCC version: ` avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0`

 - ARM GCC version: 

 - QMK Firmware version: 0.11.38

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [x] Karabiner
   - [ ] Other:

**I tried with Karabiner turned off too. No change.**

## Additional Context

drashna, 2021-01-14:
This isn't really a bug. 

Mod-Tap (and layer tap) do not work with shifted characters.  
You can only use keycodes listed here: https://docs.qmk.fm/#/keycodes_basic

EPonyA, 2021-02-10:
> This isn't really a bug.
> 
> Mod-Tap (and layer tap) do not work with shifted characters.
> You can only use keycodes listed here: https://docs.qmk.fm/#/keycodes_basic
> 
> If you want to be able to use both, you can use space cadet, as that can allow for what you want.

I had a similar issue (specifically with ``+``). This feature not working with shifted characters feels very unintuitive to me (and apparently some other people), so I personally feel it is a bug.

wkoszek, 2021-02-10:
Not sure if it helps anyone, but AFAIK I couldn't make it work via a web configurator. Regardless what combination of settings I tried then, I never made it work.

closed 2024-10-13 by drashna