qmk/qmk_firmware #11539: How to insert delays between USB reports?

zzz2496, 2021-01-13:
@drashna Hi Drashna, need a bit of help here. I have Handheld Scientific's BT-500 USB to Bluetooth adapter (http://handheldsci.com/kb/) to make my Ergodox EZ wireless. Everything is working fine except for the LT() keys. I have LT(SYMBOLS, KC_TAB) key, when HELD it will move my layer to a SYMBOLS layer, when tapped it will output TAB key. This key will report as stuck to the BT adapter. I contacted the manufacturer, and they said this:
`Thank you for purchasing our product and writing to us. Please see the QMK section of the Compatibility page (http://handheldsci.com/comp). The short answer is unfortunately we are not able to fix this on our side but it is possible to fix this on the QMK side if you have enough knowledge of it. Let us know if you have any further questions.`

After reading that document, it said something about inserting delays between USB reports. I asked if this is the same parameter as the USB polling rate, they said "NO", here's the complete reply:
`It is not the polling rate. And I don’t think it is a parameter setting in QMK. Find where multiple key reports are sent in the code and insert some delay in between. There are many keyboards supported by QMK. If you can find out which firmware this keyboard is using, perhaps we can take a look as well (provided you are able to compile and download the firmware into your kb).`

 So my best chance is to ask the QMK GURU, Professor @drashna hahahahah. Btw, they said to put delay around 10ms for each USB reports (according to the document, that is). I've been reading the QMK code, and I can't seem to find where to put the delay. Maybe you can put me into the right direction?

Thanks,

Best regards,

zzz2496

zzz2496, 2021-01-17:
Hmm, no one?

spidey3, 2021-01-17:
When your `LT(_layer_,KC_TAB)` key is pressed, QMK just stores that press. When you release it without pressing some other key, that's when QMK will actually send the press and release for `KC_TAB`. By default, there is no delay between the press and release; my guess is that your USB to Bluetooth adapter is (incorrectly) ignoring the key release event when it comes too soon after the key press.

To work around the issue, try setting
```c
#define TAP_CODE_DELAY 10
```
in your `config.h`. That will cause QMK to pause for 10 ms. between sending the press and release events to the adapter.
- If it doesn't work at 10, try increasing to a larger number. 
- If it does work at 10, try decreasing to a smaller number.

You should set this to the lowest number that works reliably.

Now that I've said all of this, I think that Handheld Scientific's explanation is avoiding the real issue, which is that their adapter is buggy. Quick press and release is a perfectly valid use case, and the adapter should handle it correctly. Of course, you probably will have a hard time getting them to fix it, so I suggest using `TAP_CODE_DELAY` as the workaround.

zzz2496, 2021-01-17:
@spidey3 Thanks for the reply, I will try that right away. Handheld Scientific loved to have QMK work on their adapter, thus I linked them to this thread. I will update this thread once I tried it.

Thanks.

zzz2496

zzz2496, 2021-01-17:
@spidey3 IT WORKS !!! Thank you very much. The magic number is indeed 10, I tried 5 to 9, it reports as stuck key, 10 works like a charm. Again, thank you very much or the guide.

zzz2496

spidey3, 2021-01-17:
> @spidey3 IT WORKS !!! Thank you very much. The magic number is indeed 10, I tried 5 to 9, it reports as stuck key, 10 works like a charm. Again, thank you very much or the guide.

No worries! Please mark the issue closed if it's no longer a problem.

closed 2021-01-17 by zzz2496

ZaxonXP, 2022-06-17:
Can this thing be changed using [Oryx ](https://configure.zsa.io/ergodox-ez/layouts/mOP6m/latest/config/)online configurator?