qmk/qmk_firmware #11543: Adding combos to my keyboard messes up some mod-taps (not the same keys)

Rougemoot, 2021-01-13:
Hi!

I've been trying to add combos to my keyboard for accented letters (`é` on `ER` for example), so I followed the tutorial in the QMK docs, and got my combos working just fine, even those that were on the same keys as mod-taps (home row mods) thanks to the [PR this article is talking about](https://precondition.github.io/home-row-mods#using-mod-taps-in-combos).

The problem that appeared then is some of my keys (mod-taps), but not all, started behaving strangely.

You can see in [my keymap](https://github.com/Rougemoot/qmk_firmware/tree/master/keyboards/jj40/keymaps/Rougemoot) the `TAB_MEH`, `ESC_FN`, and `SFT_A` (all on the base layer) are far from being the only mod-taps in my keymap, but these are the only ones being affected.

What happens is that they behave erratically, sometimes working just fine, sometimes not registering, and sometimes even freezing the whole keyboard for 5 to 10 seconds (or that may be until my frustrated key mashing triggers something else...). This for either the tap or the hold role of each key.

I've tried reducing the combos to a minimum, keeping just one and having it just `tap_keycode(KC_U);` for example, but the same behaviour persists.

The issue seems somewhat similar to #5595, in that taking the `COMBO_TERM` *seems* to make the issue happen a bit less, but I'm not ecen sure of that.

I've been trying to get help on Discord, and some very nice people have tried to help me but we couldn't find anything wrong with my keymap or config.

Any ideas what I could be doing wrong?

sevanteri, 2021-01-15:
So if you disable combos completely, the problem goes away?

Rougemoot, 2021-01-15:
Yes, disabling combos entirely makes the mod-tap issues disappear.

Maybe worth mentioning too, when using an untouched `master` branch, I have the same issue but this time with the `j` and `k` keys (`CMD_J` and `ALT_K` in my keymap), whereas with your PR (from either your `early_combos` or `combos_on_master` branch) it's `TAB_MEH`, `ESC_FN` and `SFT_A`,

I fin