qmk/qmk_firmware #11566: [Feature Request] A "layer modifier" for the QMK configurator

bad1080, 2021-01-15:
Dear QMK community,
for me as a newcomer switching between multiple layers is a daunting process, setting them up even more so. 
Because of this I wanna propose the idea of a "layer modifier", let's call it LM for right now, for the QMK configurator (https://config.qmk.fm/). While pressed LM acts like a regular modifier (it does "nothing" on its own) and in combination with a number (0-9) it switches to the according layer. Pressing LM and "3" (KC_3 and/or KC_P3) switches to layer 3 and so on.
That way layer switching uses only one key per layer ("LM" (or KC_TRNS in non-default layers) and 0-9 on a number row or num pad), you can switch from any layer to any other with an easy to set up and easy to remember combination and there is no guessing involved as to where you'll end up or how you got there.
I hope you consider this idea if at all technically possible and thanks for reading!

spidey3, 2021-01-15:
Isn't this exactly the behavior that MO(_layer_) provides?

See [Switching and Toggling Layers](https://docs.qmk.fm/#/feature_layers?id=switching-and-toggling-layers).

JonasGessner, 2021-01-23:
This can be done with #11422 (there is even an example in the docs for toggling a momentary layer), but not in qmk configurator.

bad1080, 2021-01-24:
> but not in qmk configurator.

That's why I specifically mentioned the configurator

JonasGessner, 2021-01-24:
It's not even trivially possible without the configurator currently, but #11422 makes something like what you are asking for very easy.

The configurator will only get you so far before you have to start coding if you want any kind of advanced functionality in your firmware.

bad1080, 2021-01-24:
> The configurator will only get you so far before you have to start coding if you want any kind of advanced functionality in your firmware.

Fair enough. This wasn't meant as "how can I do this" but as "feature suggestion" because I think it would simplify the process and the introduction into layers for beginners like me.

closed 2021-07-08 by stale[bot]