qmk/qmk_firmware #11572: [Bug] Volume key codes not working on Debian GNU/ Linux 10, xfce4

akmadian, 2021-01-15:
I'm using a rotary encoder to control volume level, it was working just fine on Arch linux, but when I switched to Debian, it stopped working.

I've tried recompiling and flashing from Debian, and having the encoder tap both `KC__VOLUP/KC__VOLDOWN` and `KC_VOLU/KC_VOLD` just in case, but neither worked.

I suspect this may be something to do with my kernel, but I'm not positive. I'm using the default kernel that comes with the most recent Debian netinst ISO.

## System Information

 - Keyboard:  Nullbitsco/nibble
 - Operating system: Debian GNU/ Linux 10
   - Desktop Environment: xfce4
   - `uname -a`: `Linux [hostname] 4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux`
 - AVR GCC version: 5.4.0

 - ARM GCC version: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 - QMK Firmware version: 0.10.51

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

spidey3, 2021-01-17:
This is not a QMK issue, but rather, an issue with the configuration in Linux / XFCE.

See https://unix.stackexchange.com/questions/342554/how-to-enable-my-keyboards-volume-keys-in-xfce for instructions that were [mostly] correct for me.

akmadian, 2021-01-17:
@spidey3 This worked perfectly! Thank you, I've been tearing my hair out over this : )

closed 2021-01-17 by akmadian