qmk/qmk_firmware #11649: [Feature Request] Support for Raspberry Pi Pico

ryanm101, 2021-01-21:
https://www.raspberrypi.org/blog/raspberry-pi-silicon-pico-now-on-sale/

Would be nice to support this device as an alternative to arduino

dsalaj, 2021-01-21:
I would very much like that feature too.

fauxpark, 2021-01-21:
Please see this page for information on adding support for a new MCU family: https://docs.qmk.fm/#/platformdev_selecting_arm_mcu?id=add-new-mcu-family

In short, it is non-trivial, and would require someone to commit to maintenance of the code in ChibiOS.

tzarc, 2021-05-09:
No USB support, though.

siddacious, 2021-08-10:
I'll take a look at this; I just got some rp2040 pro micros that I'm itching to use

hanya, 2021-09-21:
We have USB driver in ChibiOS-Contrib now.

Agilulfo, 2021-11-06:
https://github.com/qmk/qmk_firmware/pull/14208 was recently merged.
Does it mean that this issue is now solved? :smile:

rhruiz, 2021-11-06:
Hey seem to have merged 20.3 in not 21.6. And 20.3 does not seem to have RP2040 support yet, can anyone confirm that?

tzarc, 2021-11-06:
21.6.x was canceled by ChibiOS itself.
This PR has to wait for 21.11.x to get released, and due to the timing will not make this breaking changes cycle.

gompa, 2021-11-08:
> 21.6.x was canceled by ChibiOS itself. This PR has to wait for 21.11.x to get released, and due to the timing will not make this breaking changes cycle.

i think 21.11.x released on the 7th 
https://forum.chibios.org/viewtopic.php?f=7&t=5963

tzarc, 2021-11-08:
It's a release candidate.

[image]

tzarc, 2021-11-08:
Also, ChibiOS-Contrib is not yet tracking 21.11.x, so there's no path for upgrade regardless.

charlie-hotel, 2021-12-12:
Looks like 21.1.1 is out now.

tzarc, 2021-12-12:
> Looks like 21.1.1 is out now.

Of ChibiOS, yes. Of ChibiOS-Contrib, no.

hamano, 2021-12-20:
I was waiting rp2040 support for qmk as well as you.
Now, I question the necessity of ChibiOS support, since I found sekigon's branch that works very well with rp2040.
https://github.com/sekigon-gonnoc/qmk_firmware/tree/rp2040

You can try example/rp2040_example and example/rp2040_split with rp2040.
It works with pico-sdk, and the split keyboards communicate using PIO.
As you already know, the biggest advantage of Pico is PIO. Is it possible to use PIO on ChibiOS?
Dou you have interest to be able to do matrix scanning using PIO instead of CPU?
Would you consider merging sekigon's branch or linking pico-sdk instead of ChibiOS?
Thanks.

tzarc, 2021-12-21:
>  Would you consider merging sekigon's branch or linking pico-sdk instead of ChibiOS?

Given that the branch is quite out of date compared to QMK and doesn't follow the structural patterns, it'd require a fair bit of work to do so. ChibiOS is still preferable at this point, as we're not having to rely on a single person making changes or bugfixes.

tzarc, 2021-12-21:
This commit, for instance, goes in the opposite direction of QMK:
https://github.com/sekigon-gonnoc/qmk_firmware/commit/838055cf9edda32033d6745294844809202074b9

hanya, 2021-12-22:
ChibiOS uses pico-sdk to support RP2040. So you can call some functions from pico-sdk in QMK code.

hamano, 2021-12-23:
I believe that ChibiOS is the better choice if PIO is available.
Thanks.

zykrahgaming, 2022-01-10:
Just to update anyone still interested regarding QMK RP2040 support, check out https://github.com/ChibiOS/ChibiOS-Contrib/pull/310. From what I know, work on getting preliminary RP2040 support in QMK cannot happen until chibios-contrib maintainers merge this PR.

ankitsxchdeva, 2022-01-29:
Thank you for linking that PR! Looks like there's progress being made on it. Probably a stupid question but would that ChibiOS PR allow all RP MCUs to be QMK compatible?

zykrahgaming, 2022-01-29:
> Probably a stupid question but would that ChibiOS PR create the possibility of all RP MCUs to be QMK compatible?

As far as it wont add support for existing MCUs (ie on other raspis), but it could make adding future RP MCUs easier if they ever do come out in the future.

nodupe, 2022-02-06:
I'm happy to see that the commit on Chibios-contrib was merged!

maberalc, 2022-02-10:
Will this lead to the support finally? Or it is already supported as it was merged???

KarlK90, 2022-02-10:
> Will this lead to the support finally? Or it is already supported as it was merged???

https://github.com/qmk/qmk_firmware/pull/14877 is the tracking PR for RP2040 support in QMK. It is actively being worked on but no ETA yet.

gril002, 2022-07-03:
Congrats to everyone involved on merging #14877 to everyone involved. I can see it's now part of the develop branch. When can we expect the full release and what else need to be done before? Or is the develop branch stable enough to use?

tzarc, 2022-07-03:
`develop` is usually pretty stable, so should be fine for use.
Otherwise, the next merge of `develop` into `master` will occur on August 28.

closed 2022-07-03 by tzarc

xbliss, 2022-08-26:
> `develop` is usually pretty stable, so should be fine for use. Otherwise, the next merge of `develop` into `master` will occur on August 28.

Is this still on schedule or have the dates moved?

kaerbr, 2022-10-27:
I guess its now available via https://config.qmk.fm/#/adafruit/macropad/LAYOUT (: