qmk/qmk_firmware #11707: [Feature Request] Encoder can be "held" down, and there should be and option to stop this.

Kycklingris, 2021-01-27:
## Feature Request Type

- [ ] Alteration (enhancement/optimization) of existing feature(s)

## Description

I don't really know if this should be called a bug or not, but I have a "ACZ11BR4E-15FD1-20C" rotary encoder connected to a Atmega 32A, and well, sometimes it gets kind of stuck, jumping to 0 or 100 in a couple of seconds. Now, I don't really know if it would even be possible, if I knew, I probably would have tried to fix it myself, but if qmk could just remember if the encoder is still connected and not send the volume keycode, that would be great. I do realize it is entirely possible it's just my encoder that's the problem.

Oh well, thank you for taking the time to read this and, thank you to everyone working on qmk. My keyboard wouldn't have even 10% of the features it has now otherwise.

fauxpark, 2021-01-27:
Have you tried adding this to your config.h:
```c
#define TAP_CODE_DELAY 10
```

Kycklingris, 2021-01-27:
Been a while since the last time a worked on my keyboard, but looking at it now, I can't find a TAP_CODE_DELAY,  I also can't find it in the docs, but I do have TAPPING_TERM 200. Although I don't know if they are related. If I remember correctly that is the time delay between a tap and a hold no? I'll test it with it, but if I remember correctly I have tried something like that before.

fauxpark, 2021-01-27:
If you are calling `tap_code()` or `tap_code16()` in your `encoder_update()` function, the default delay between sending the keydown and keyup events is 0ms (effectively immediately). So there is a possibility that one of those events may not be registered correctly on the host side, which can lead to what you are experiencing.

Defining a `TAP_CODE_DELAY` of about 5-10ms should give enough time for the OS to be able to distinguish the two events.

Kycklingris, 2021-01-27:
Well, seems to works so far, with my tiny amount of testing. Now I just need to start another issue, although I'm sure that isn't a bug. 

This should probably be added to the docs tho.
Thank you so much.

closed 2021-01-27 by Kycklingris