qmk/qmk_firmware #11708: Help wanted with usb something or other, I don't really know the cause of the problem. And maybe other, explained at the bottom.

Kycklingris, 2021-01-27:
Well then, I am very much new to using github outside of just my own repositories, and I have very recently gotten help with 1/3 of my issues. So if I'm doing something wrong, do tell me.

Soo, the problem, and the last two I am guessing they aren't bugs, but rather my own fault, and even worse, one of them may actually not even be a problem with qmk....

1. the one that may or may not be a problem that comes from qmk, is, my keyboard has a very hard time to get connected to multiple different computers, I know this isn't driver related at least. It may take multiple tries of plugging it in and out, and spamming reset and such. As far as I know the only error code it has given is "Windows has stopped this device because it has reported problems. (Code 43) A USB port reset request failed." 

My keyboard is based on coseyfannituttis discipline [](url)https://github.com/coseyfannitutti/discipline  and my own pcb design [](url)https://oshwlab.com/malte.linde.neveling/discipline#P3
Now, I'm not expecting help with it if the problem comes from the pcb, but it shouldn't even be the problem as I kept almost all usb components the same as the original discipline.
Second possible problem source would probably be the bootloader, and it should also be the same one that the discipline uses found here [](url)https://github.com/coseyfannitutti/USBaspLoader/tree/atmega32a  So I don't expect it to be the problem either, but I have barely any experience in the hardware side of programming.
Lastly, it could be qmk, and well, I have no idea there [](url)https://github.com/Kycklingris/keyboard

Now, I have no idea if I should be creating another issue for the second problem, so I wont write about it in detail, unless asked to, but basically, I have followed all the tutorials about the rgb matrix, and found several issues, and done them, but the light backlight still wont turn off when my pc is off. Although I guess it could probably be related to the issue above.

spidey3, 2021-01-29:
This might be another instance of Issue #11667 or #11389...

Kycklingris, 2021-01-29:
Well the first one does indeed seem quite similar, I did try today to update qmk to the latest version, but I'll try to downgrade with the same one tomorrow. Well regardless of if it works, thank you.

MajorKoos, 2021-02-10:
Try adding this to your rules.mk file:
SHARED_EP_ENABLE = yes
KEYBOARD_SHARED_EP = yes

Or you could try building against 0.10.54 - just before the additional endpoint was added.

Kycklingris, 2021-02-10:
I almost completely forgot I asked here, so thank you for still answering. But 

> Try adding this to your rules.mk file:
> SHARED_EP_ENABLE = yes
> KEYBOARD_SHARED_EP = yes
> 

Seems to have worked so far, although I have only tried replugging it a couple of times and pressing reset a couple of times. I still haven't tested if it changes anything when I turn off the computer tho. But thank you so much for your help @MajorKoos !

MajorKoos, 2021-02-10:
You *may* have issues in the bios with that config, but it depends on your bios.
I've got similar issues on a bunch of PCBs using the atmega32a clocked at 12MHz.
Building against 10.54 is more stable than using the shared endpoint with 11.x IMO, but I think it depends on your setup.

Kycklingris, 2021-02-11:
Well it's not like I go into the bios often anyway. This makes it so much easier to use the keyboard. Once again, thank you so much.

Now the only problem I have is the rgb matrix not turning off. It is late for me though, so I'll check again tomorrow if it's just that I commented out the rgb off on sleep line or something

spidey3, 2021-02-11:
> Now the only problem I have is the rgb matrix not turning off.

In what context? When the computer sleeps? Or when you use a key to disable it?

spidey3, 2021-02-11:
If it's about while the computer is sleeping, try defining `#define RGB_DISABLE_WHEN_USB_SUSPENDED` in your `config.h`.

Kycklingris, 2021-02-11:
Right, the `#define RGB_DISABLE_WHEN_USB_SUSPENDED true` and `void suspend_power_down_kb(void) {
    rgb_matrix_set_suspend_state(true);
    suspend_power_down_user();
}

void suspend_wakeup_init_kb(void) {
    rgb_matrix_set_suspend_state(false);
    suspend_wakeup_init_user();
}`
in keyboard.c doesn't seem to work, I have tried both the keyboard.c and keymap variant of it, but neither works. It also shouldn't be a typo or anything as I copied it from the docs. 

And, when I say it doesn't turn off, I do mean when the computer sleeps.

spidey3, 2021-02-11:
Are there any hubs, etc. between your keyboard and your computer?
I'm wondering whether the keyboard is actually being suspended...

Kycklingris, 2021-02-11:
No, it's a direct connection. But there is the horrible possibility that the hardware is wrong...
It shouldn't be the problem, but as I did it largely to learn, I wouldn't really know.

spidey3, 2021-02-11:
Oh - wait - the MCU (atmega32a) on discipline uses vusb; it is a known issue that suspend_power_down / suspend_wakeup_init are never called on vusb. It's something we plan to work on shortly.

Kycklingris, 2021-02-11:
Okay, thank you so much for your help! Then I guess I'll need to keep a lookout on when it gets updated. Nice to know I didn't use any money on a project that can't be fixed :)

I don't really know, but I guess I should close it?

spidey3, 2021-02-11:
Can you try the code in this branch: https://github.com/spidey3/qmk_firmware/tree/fix/vusb_suspend
It's based on the latest in `develop`, but also includes a simple try at making suspend work on vusb. 
Note, I haven't tested it - as I don't have any board handy that uses vusb...

Kycklingris, 2021-02-11:
Sure I'd be glad to help!

Soo, I tried to install the fork you linked in two different ways. 1, by just overwriting the files inside the qmk firmware directory and 2, just removing the entire directory and replacing it. To then run qmk setup again. But both times gave me 
`Error: C:/Users/malte.lindeneveling/qmk_firmware is too old or not set up correctly!
Please update it or remove it completely before continuing.
`
I also got it when I tried to use qmk flash regardless.

spidey3, 2021-02-11:
I would have done:
`git remote add spidey3 git@github.com:spidey3/qmk_firmware.git`
`git checkout spidey3/fix/vusb_suspend`
`make git-submodule`
`qmk compile...`

Kycklingris, 2021-02-11:
Right, so I tried the first one, but it gave me 
`$ git remote add spidey3 git@github.com:spidey3/qmk_firmware.git
fatal: not a git repository (or any of the parent directories): .git
`
(Hmmm, I probably need to learn more about git before markdown tho)
Welp, I realised I needed an ssh key. But after that and adding the remote worked, when I tried to checkout it gave me 
`$ git checkout spidey3/fix/vusb_suspend
error: pathspec 'spidey3/fix/vusb_suspend' did not match any file(s) known to git
`
This one I have no idea how to even try to figure out...

spidey3, 2021-02-11:
Can we switch to Discord for this discussion?
https://discord.gg/pqr8ESAz

Kycklingris, 2021-02-11:
Sure, which channel?

spidey3, 2021-02-11:
Meet me in #firmware

closed 2021-07-08 by stale[bot]