qmk/qmk_firmware #11709: [Feature Request] GFX standardization

M1Sports20, 2021-01-27:
I have a custom keyboard with an SH1106 connected for 4 wire SPI.  I have a couple of questions
1.  I seen ugfx included in the repo, I was going to just use this since its nice to have project support.  However, browsing tickets it seems this is going away.  Is this true? Why is it going away?
2. I am willing to modify the SSD1306 library to support SPI, but didn't want to put the effort in if there are better longer term solutions.
3. Why don't we use a project like [u8g2](https://github.com/olikraus/u8g2).  We don't have to maintain a bunch of drivers with configuration options?  I am willing to try and pull it in, but wanted to see if this was a way the project wanted to go?

I am going to get my display working and am trying to help the project while selecting the best maintainable solution;.

tzarc, 2021-01-27:
Reasoning:
* uGFX is deprecated due to licensing incompatibilities regarding distribution
* u8g2 only does monochrome

Fear not; this is being mostly addressed in #10174. A backend driver would need to be written for the OLEDs properly (it currently just wraps existing functionality as a quick "win"), but it's intentionally designed to allow for pluggable backends whilst leveraging standard drawing routines where applicable.

Font rendering is the last hurdle before it goes for final review.

M1Sports20, 2021-01-27:
Alright, wonderful.  It looks like it has a high chance of going mainline.  I think I will checkout your branch and see what I can do on the backend.  I think I have a way of testing both SSD1306(over I2c) and SH1106(over SPI).

closed 2021-01-27 by M1Sports20