qmk/qmk_firmware #11710: [bug] Installing QMK via MINGW64 errors out halfway through

acpearson, 2021-01-27:
## Describe the Bug
[image]
[image]
[image]

I was following the install instructions for windows when these errors popped up. I am very new to this, please be gentle.

Thank you
## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2021-01-27:
You need to run this command first:
```
pacman --needed --noconfirm --disable-download-timeout -S git mingw-w64-x86_64-toolchain mingw-w64-x86_64-python3-pip
```
If it errors out, post it here.

acpearson, 2021-01-27:
I did. That ran successfully.
[image]

fauxpark, 2021-01-27:
Are you sure? That's not the full output.

acpearson, 2021-01-27:
Is this the full output?
[image]
[image]
[image]
[image]

fauxpark, 2021-01-27:
That doesn't look like the same command I posted...

acpearson, 2021-01-27:
You're right! I didn't see that I forgot a little bit. Let me try again.

acpearson, 2021-01-27:
Looks like that fixed the problem!! Thank you!

closed 2021-01-27 by acpearson