qmk/qmk_firmware #11712: Can STM32F103 be set as configured as stm32f303？

Weirdo-F, 2021-01-27:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
Today, try to use STM32F103 to design keyboard.

After looking at several keyboard codes using this MCU, I found that the configuration is troublesome. I don't know if STM32F103 can be configured as simply as stm32f303.

tzarc, 2021-01-27:
Using the `develop` branch is much simpler, e.g.: [practice60](https://github.com/qmk/qmk_firmware/tree/develop/keyboards/cannonkeys/practice60).

Weirdo-F, 2021-01-28:
> Using the `develop` branch is much simpler, e.g.: [practice60](https://github.com/qmk/qmk_firmware/tree/develop/keyboards/cannonkeys/practice60).

thanks！

closed 2021-01-30 by Weirdo-F